CadQuery/cadquery #1706: Can CadQuery detect collision among parts?

davidedelvento, 2024-11-18:
First, I'm new to CAD (but not to programming). I find the OpenSCAD approach (especially when extended by BOSL2) very attractive and intuitive. However there is an important missing feature which is driving me crazy (given the design mistakes I am making), and therefore I looked for alternatives, but have found none that does what I need/want, even though CQ might (not clear to me from the documentation, hence this ticket).

For the sake of the discussion, imagine that I am designing a number of complex-shaped moving parts, attached to one another with hinges placed on arbitrary places. Something like the https://en.wikipedia.org/wiki/Double_pendulum but with many more parts, all of various shapes (at the moment all the hinging is happening on parallel lines, exactly like in the double pendulum, but it would be great if they could be modeled arbitrarily). Unlike the double pendulum (which has its arms in different planes and can't collide with each other), my parts can collide with each other, and the range of movements are therefore restricted by such collisions. I do **not** need a physics modeling, just detect collisions among parts.

I know I can pre-emptively consider every possible collision, and design accordingly. For a (simpler) example, even in plain OpenSCAD I can design a square or rectangular peg of the appropriate size to fit in a round hole. However I can do that only if I beforehand identify the possible collisions and design on purpose to avoid them. If I make a mistake and/or overlook something, OpenSCAD is not going to help me much (its animations features do help a tiny bit, but really tiny). I see that CQ is much more powerful in this regard with Assemblies and Constrains, but even so, it feels like I should define possible collision points on my own beforehand, which I would like to avoid, given the complexity of the shapes and the range of movement involved.

So can CQ help me with that? Or should I look at something that has a physics engine, even though I don't really need that? If the latter, which one you'd suggest me looking into? I've seen BowlerStudio and BRL-CAD but neither seem particularly intuitive as OpenSCAD or CQ.

Thanks a lot in advance!

adam-urbanczyk, 2024-11-18:
How about using `intersect` on the two suspect parts? Otherwise something like this on the OCP/OCCT level: https://dev.opencascade.org/content/collision-detection-working

davidedelvento, 2024-11-19:
Thanks, those look like good option. I will explore both approaches (the second looks a bit intimidating but perhaps it's just because I'm not familiar with the thing yet)