CadQuery/cadquery #1707: do cskHole and cboreHole only work along z-axis?

bigfighter, 2024-11-20:
This works like a charm: `cq.Workplane('XY').box(10, 10, 10).faces('>Z').cskHole(1, 2, 90)`
But when I switch to a different face, something peculiar happens:
[image]
Same for `.cboreHole(1, 2, 2)` instead of `.cskHole(1, 2, 90)`
Is there a way to drill sideways?

No big deal, I can always union a cone and a cylinder (or two cylinders for countersunk), move them into position and subtract them.
But cboreHole/cskHole would be so much more convenient.

jmwright, 2024-11-20:
@bigfighter You need to establish a workplane after selecting the face.

```python
import cadquery as cq

box = cq.Workplane().box(100, 100, 100)

box = box.faces(">Y").workplane().cskHole(10, 20, 90)

show_object(box)
```
[image]

closed 2024-11-20 by jmwright

bigfighter, 2024-11-20:
beautiful, thank you!