CadQuery/cadquery #1709: Hole not at the current workplane

davidedelvento, 2024-11-23:
The following

```python
a = Workplane("XY")
a = a.box(9, 20, 4)
a = a.faces("<Y")
a = a.workplane()
a = a.rect(4,4).extrude(13)
a = a.faces(">X")
a = a.hole(2)
```

generates

[image]

in which the hole is rotated 90 degrees from the current workplane.

Is this a bug or am I doing something incorrect?

adam-urbanczyk, 2024-11-23:
You changed the workplane here `a = a.faces("<Y")`.  So the hole direction looks OK I think.

adam-urbanczyk, 2024-11-23:
Are you maybe confusing workplane and selection?

davidedelvento, 2024-11-23:
> Are you maybe confusing workplane and selection?

Possibly. But https://cadquery.readthedocs.io/en/latest/classreference.html#cadquery.Workplane.hole says

> The surface of the hole is at the current workplane.

Arguably "the surface" is ambiguous but I understood it as the "surface" that one would drill through during the manufacturing.

Can you please clarify?

davidedelvento, 2024-11-24:
Okay so adding a `workplane` fixes the issue, as below

```python
a = Workplane("XY")
a = a.box(9, 20, 4)
a = a.faces("<Y")
a = a.workplane()
a = a.rect(4,4).extrude(13)
a = a.faces(">X").workplane()
a = a.hole(2)
```

I am probably still a bit confused between workplane and selection, but at least I can move forward and learn.