CadQuery/cadquery #1711: Can't export Workplanes

davidedelvento, 2024-11-24:
Following the example from https://github.com/CadQuery/cadquery/blob/2629bf94900a6acedd00b8b50f00c41a30b4785c/doc/importexport.rst#exporting-step and the documentation at https://cadquery.readthedocs.io/en/latest/classreference.html?highlight=export#cadquery.Workplane.export

[image]
(note it says version 2.4 on the left)

## To Reproduce

```python
$ python
Python 3.10.12 (main, Nov  6 2024, 20:22:13) [GCC 11.4.0] on linux
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery
>>> cadquery.__version__
'2.4.0'
>>> box = cadquery.Workplane().box(10, 10, 10)
>>> box.export("box.step")
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Workplane' object has no attribute 'export'. Did you mean: 'exportSvg'?

```

## Backtrace

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
AttributeError: 'Workplane' object has no attribute 'export'. Did you mean: 'exportSvg'?

```

## Environment

OS: Linux

```
Distributor ID:	Ubuntu
Description:	Ubuntu 22.04.5 LTS
Release:	22.04
Codename:	jammy
```

Was CadQuery installed using Conda?: No, with pip. Note that I had to jump through the hoops detailed in #1626 but that seems unlikely to cause this issue?

Output of `pip list` from your active virtual environment:
```
Package                       Version
----------------------------- --------------
aiohappyeyeballs              2.4.3
aiohttp                       3.11.7
aiosignal                     1.3.1
alabaster                     1.0.0
arrow                         1.3.0
astroid                       3.3.5
asttokens                     2.4.1
async-timeout                 5.0.1
asyncssh                      2.18.0
atomicwrites                  1.4.1
attrs                         24.2.0
autopep8                      1.5.5
babel                         2.16.0
backports.tarfile             1.2.0
beautifulsoup4                4.12.3
binaryornot                   0.4.4
black                         24.10.0
bleach                        6.2.0
cadquery                      2.4.0
cadquery-ocp                  7.7.2
casadi                        3.6.7
certifi                       2024.8.30
cffi                          1.17.1
chardet                       5.2.0
charset-normalizer            3.4.0
click                         8.1.7
cloudpickle                   3.1.0
comm                          0.2.2
cookiecutter                  2.6.0
CQ-editor                     0.3.0.dev0
cryptography                  43.0.3
debugpy                       1.8.9
decorator                     5.1.1
defusedxml                    0.7.1
Deprecated                    1.2.15
diff-match-patch              20241021
dill                          0.3.9
docstring-to-markdown         0.15
docutils                      0.21.2
exceptiongroup                1.2.2
executing                     2.1.0
ezdxf                         1.3.4
fastjsonschema                2.20.0
flake8                        3.8.4
fonttools                     4.55.0
frozenlist                    1.5.0
idna                          3.10
imagesize                     1.4.1
importlib_metadata            8.5.0
inflection                    0.5.1
intervaltree                  3.1.0
ipykernel                     6.29.5
ipython                       8.29.0
isort                         5.13.2
jaraco.classes                3.4.0
jaraco.context                6.0.1
jaraco.functools              4.1.0
jedi                          0.17.2
jeepney                       0.8.0
jellyfish                     1.1.0
Jinja2                        3.1.4
jsonschema                    4.23.0
jsonschema-specifications     2024.10.1
jupyter_client                8.6.3
jupyter_core                  5.7.2
jupyterlab_pygments           0.3.0
keyring                       25.5.0
Logbook                       1.8.0
markdown-it-py                3.0.0
MarkupSafe                    3.0.2
matplotlib-inline             0.1.7
mccabe                        0.6.1
mdurl                         0.1.2
mistune                       3.0.2
more-itertools                10.5.0
multidict                     6.1.0
multimethod                   1.9.1
mypy-extensions               1.0.0
nbclient                      0.10.0
nbconvert                     7.16.4
nbformat                      5.10.4
nest-asyncio                  1.6.0
nlopt                         2.7.1
nptyping                      2.0.1
numpy                         1.23.5
numpydoc                      1.8.0
packaging                     24.2
pandocfilters                 1.5.1
parso                         0.7.0
path                          17.0.0
pathspec                      0.12.1
pexpect                       4.9.0
pickleshare                   0.7.5
pip                           24.3.1
platformdirs                  4.3.6
pluggy                        1.5.0
prompt_toolkit                3.0.48
propcache                     0.2.0
psutil                        6.1.0
ptyprocess                    0.7.0
pure_eval                     0.2.3
pycodestyle                   2.6.0
pycparser                     2.22
pydocstyle                    6.3.0
pyflakes                      2.2.0
PyGithub                      2.5.0
Pygments                      2.18.0
PyJWT                         2.10.0
pylint                        3.3.1
pylint-venv                   3.0.4
pyls-black                    0.4.7
pyls-spyder                   0.4.0
PyNaCl                        1.5.0
pyparsing                     3.2.0
PyQt5                         5.12.3
PyQt5-Qt5                     5.15.15
PyQt5_sip                     12.15.0
pyqtgraph                     0.13.7
PyQtWebEngine                 5.12.1
PyQtWebEngine-Qt5             5.15.15
python-dateutil               2.9.0.post0
python-jsonrpc-server         0.4.0
python-language-server        0.36.2
python-lsp-black              2.0.0
python-lsp-jsonrpc            1.1.2
python-lsp-server             1.12.0
python-slugify                8.0.4
pytoolconfig                  1.3.1
pyuca                         1.2
pyxdg                         0.28
PyYAML                        6.0.2
pyzmq                         26.2.0
QDarkStyle                    3.0.2
qstylizer                     0.2.4
QtAwesome                     1.3.1
qtconsole                     5.6.1
QtPy                          2.4.2
referencing                   0.35.1
requests                      2.32.3
rich                          13.9.4
rope                          1.13.0
rpds-py                       0.21.0
Rtree                         1.3.0
SecretStorage                 3.3.3
setuptools                    59.6.0
six                           1.16.0
snowballstemmer               2.2.0
sortedcontainers              2.4.0
soupsieve                     2.6
Sphinx                        8.1.3
sphinxcontrib-applehelp       2.0.0
sphinxcontrib-devhelp         2.0.0
sphinxcontrib-htmlhelp        2.1.0
sphinxcontrib-jsmath          1.0.1
sphinxcontrib-qthelp          2.0.0
sphinxcontrib-serializinghtml 2.0.0
spyder                        5.0.0
spyder-kernels                2.0.5
stack-data                    0.6.3
superqt                       0.6.7
tabulate                      0.9.0
text-unidecode                1.3
textdistance                  4.6.3
three-merge                   0.1.1
tinycss2                      1.4.0
toml                          0.10.2
tomli                         2.1.0
tomlkit                       0.13.2
tornado                       6.4.2
traitlets                     5.14.3
types-python-dateutil         2.9.0.20241003
typing_extensions             4.12.2
typish                        1.9.3
ujson                         5.10.0
urllib3                       2.2.3
watchdog                      1.0.2
wcwidth                       0.2.13
webencodings                  0.5.1
whatthepatch                  1.0.7
wrapt                         1.17.0
wurlitzer                     3.1.1
yapf                          0.43.0
yarl                          1.18.0
zipp                          3.21.0
```

Using: nothing other than cadquery itself

davidedelvento, 2024-11-24:
Probably obvious to the developers, but for the sake of other users' sanity: here is a possible workaround (tested only minimally)

```python
from cadquery import exporters
exporters.export(box, "box.step")
```

lorenzncode, 2024-11-24:
>(note it says version 2.4 on the left)

It is incorrect.   

For now, if you are on 2.4 rather than latest master branch, switch to "stable" docs version (select in bottom right). 
[image]

adam-urbanczyk, 2025-02-06:
AFAICT solved in 2.5

closed 2025-02-06 by adam-urbanczyk