CadQuery/cadquery #1712: Are there any approaches to combine/union the individual letters together in the cq.Workplane().text("Hello, CadQuery") object?

huskier, 2024-11-25:
We've made 3D texts with the following code. We've tried several methods to combine/union the individual letters together; however, we always obtain the individual letters in the STEP structure tree as the attached image. Actually, we just want to obtain a single branch in the STEP structure tree as in the attached image, not the 16 part branches as shown.

```
import cadquery as cq
from cadquery.vis import show

text_string = "Hello, CadQuery!"
font = "Arial"
size = 10  # in millimeters
text3D = cq.Workplane("XZ").text(text_string, combine = True, font = font, fontsize = size, distance = 2) 
text3D = text3D.union(text3D, glue = True)
text3D = text3D.combine(text3D)
text3D = text3D.val()
text3D = text3D.fuse(text3D)
cq.exporters.export(text3D, "text3D.step", exportType = "STEP")
```

[image]

jmwright, 2024-11-28:
I'm not sure if it will work, but you can put the text into the top level of an assembly and then try a [fused export](https://cadquery.readthedocs.io/en/latest/importexport.html#fused).

adam-urbanczyk, 2024-11-28:
Ask FreeCAD, your `text3D` is already a compound.

```
In[34]: text3D.objects

Out[34]: [<cadquery.occ_impl.shapes.Compound at 0x1ed45303680>]
```

huskier, 2024-12-08:
Thanks for both of you, @jmwright @adam-urbanczyk 

Perhaps for discrete objects without any connections, they cannot be united together.

closed 2024-12-08 by huskier