CadQuery/cadquery #1714: CadQuery via pip is Broken Out of the Box!

dani-corie, 2024-11-27:
Hi!

I have first installed cadquery via pip in an existing venv, and it crashed on import.
Then I have created a *new venv*, and installed cadquery via pip (`pip install cadquery`), and it still crashed on import.

```
Traceback (most recent call last):
  File "/home/dani/Workshop/pyAtelier/tech_pocs/cadquery_test.py", line 1, in <module>
    import cadquery
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/cadquery/__init__.py", line 22, in <module>
    from .occ_impl import exporters
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/cadquery/occ_impl/exporters/__init__.py", line 10, in <module>
    from ...cq import Workplane
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/cadquery/cq.py", line 62, in <module>
    from .sketch import Sketch
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/cadquery/sketch.py", line 28, in <module>
    from .occ_impl.sketch_solver import (
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/cadquery/occ_impl/sketch_solver.py", line 3, in <module>
    from nptyping import NDArray as Array
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/nptyping/__init__.py", line 30, in <module>
    from nptyping.ndarray import NDArray
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/nptyping/ndarray.py", line 32, in <module>
    from nptyping.shape_expression import (
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/nptyping/shape_expression.py", line 35, in <module>
    from nptyping.typing_ import Literal
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/nptyping/typing_.py", line 66, in <module>
    Bool8: TypeAlias = np.bool8
  File "/home/dani/Workshop/.venv-3.10-cadquery/lib/python3.10/site-packages/numpy/__init__.py", line 414, in __getattr__
    raise AttributeError("module {!r} has no attribute "
AttributeError: module 'numpy' has no attribute 'bool8'. Did you mean: 'bool'?
```

Of course I know by now what happened: #1626 #1616 #1617 etc. I am aware the fix is in `master` by now.

But CadQuery is still broken in pip. Keeping numpy back by a major version would be pretty annoying, but if at least it was in the dependencies to pip, it would just mean keeping a separate venv. This way, installing via pip from scratch is close to impossible for people who aren't software engineers.

I see you have missed your June release. I get it, I don't want to be that person who makes demands and bangs the table at volunteer devs at FOSS projects... I imagine you're all using conda, and pip being broken is a minor stain. But it means that for any newbie coming to your project, your project looks broken.

Could you at least pull the pip package from the repo and just say 'install via conda pls until next release', or push an update with correct dependencies? And I'm excited for the belated June release... or will it be January now? ;)

adam-urbanczyk, 2024-11-28:
Conda is the first installation method mentioned in the docs and in the README not without a reason. It is really not that hard to get a conda env up and running ( https://cadquery.readthedocs.io/en/latest/installation.html#install-the-conda-package-manager ).

 You can also install master via pip (`pip install git+...`), though last time the issue was with the nlopt wheel (which is a dep of CQ). Release is coming "soon", but a release does not imply working wheels, especially if 3rd party wheels would cause issues.

BTW: you can read extensively about the issues with pip here https://pypackaging-native.github.io/

roipoussiere, 2024-12-21:
Same problem with Poetry, I was able to make it work with:

    poetry add multimethod typish nlopt casadi git+https://github.com/CadQuery/cadquery.git

jmwright, 2025-01-07:
CadQuery 2.5.2 is available and this should be fixed.

closed 2025-01-07 by jmwright