CadQuery/cadquery #1722: Bending an extruded surface

lenianiva, 2024-12-04:
If I have a sketch and extrude a solid from the sketch, is it possible to bend the extruded solid as if it was fed to a rolling machine?

I don't have a code example because without this bending operation I don't know how to model this effect.

e.g. https://blenderartists.org/t/bendind-in-a-plane-into-a-cilinder/1476118

adam-urbanczyk, 2024-12-06:
No, not right now. Do you want to map the pattern onto a face?

lenianiva, 2024-12-06:
> No, not right now. Do you want to map the pattern onto a face?

I want to map the pattern onto a cylinder and cut out a hole in the cylinder

adam-urbanczyk, 2024-12-06:
If you have a 2D array of things, you could map them to the parametric space of the cylinder, position and project onto it. Take a look here: https://github.com/CadQuery/cadquery/blob/2726421dee5852fdd689a8b286fec6271ed968b1/cadquery/occ_impl/shapes.py#L5178 it should be few lines of code actually

lenianiva, 2025-03-30:
> If you have a 2D array of things, you could map them to the parametric space of the cylinder, position and project onto it. Take a look here:
> 
> [cadquery/cadquery/occ_impl/shapes.py](https://github.com/CadQuery/cadquery/blob/2726421dee5852fdd689a8b286fec6271ed968b1/cadquery/occ_impl/shapes.py#L5178)
> 
> Line 5178 in [2726421](/CadQuery/cadquery/commit/2726421dee5852fdd689a8b286fec6271ed968b1)
>  rv.append(f.project(base, f.normalAt())) 
> it should be few lines of code actually

but I don't have a 2d array. instead I have a sketch that I extrude out to form the pattern.

In terms of construction, I laser etch the pattern on a sheet of metal and then bend it in a rolling machine into a cylinder.