CadQuery/cadquery #1723: Sweep Path Orientation Issue with Parametric Wave Path

HamzaHassan9320, 2024-12-04:
When sweeping a rectangular profile along a parametric wave path, the profile's orientation changes incorrectly as the sweep progresses. This happens regardless of whether `isFrenet` is set to `True` or `False`. The sweep initially aligns with the path correctly but distorts as it continues, making the result inconsistent.

**Code Example**
Below is a minimal, complete example to reproduce the issue. This can be copied and run directly in CQ-Editor:

`import cadquery as cq
import math

# Parameters for the wave spring
t1 = 0  # Start parameter
t2 = 50  # End parameter
num_points = 500  # Number of points for smoothness
radius = 50
(width, thickness) = (4, 2)

# Generate the wave spring path
points = []
delta_t = (t2 - t1) / num_points
for i in range(num_points + 1):
    t = t1 + i * delta_t
    # Parametric equations for the wave spring
    x = radius * math.sin(t)
    y = radius * math.cos(t)
    z = 7 * math.sin(3.5 * t) + 3 * t
    points.append((x, y, z))

# Create the wave path as a spline
wave_path = cq.Workplane("XY").spline(points)

# Define a rectangular profile and sweep along the path
result = (
    cq.Workplane("YZ")  # Profile aligned to XZ plane
    .center(radius, 0)  # Offset to the radius of the wave spring
    .rect(width, thickness)  # Create a rectangle profile
    .sweep(wave_path, isFrenet=False)  # Sweep along the wave path
)`

**Observed Behavior**
The sweep starts with the profile aligned correctly to the path.
As the path progresses, the profile's orientation distorts and deviates from the intended alignment.

**Expected Behavior**
The rectangular profile should maintain consistent orientation along the entire wave path, ensuring a uniform sweep.

**Screenshots**
Sweep with normal=(0,0,1): 
[image]

Sweep with no normal: 
[image]

Sweep with a circular profile (for comparison)(best to understand how it changes with the path)
[image]

HTeyssedre, 2024-12-10:
Same situation I think, sweeping a wire profile along a spline wave does not work properly (or as expected). 
Minimal example:

import cadquery as cq
import numpy as np

support = cq.Workplane("XY").rect(50, 100).offset2D(5)
nodes = [support.consolidateWires().val().positionAt(t) + cq.Vector(0, 0, 2+2*np.sin(20*np.pi*t)) for t in np.linspace(0,1,100)]
sweepingPath = cq.Workplane("XY").polyline(nodes).close() # or .spline

ref0 = cq.Vector(nodes[0].y , nodes[0].z)
points = [ ref0 + cq.Vector(5, 0),
          ref0 + cq.Vector(-5, 0),
          ref0 + cq.Vector(-5, -4)
        ]

profile = cq.Workplane("YZ", origin=(nodes[0].x,0,0)).polyline(points).consolidateWires() 
sweptPlane = profile.sweep(sweepingPath, isFrenet=False)

show_object(sweptPlane)

[image]
[image]
(nodes).close()

HamzaHassan9320, 2024-12-10:
@adam-urbanczyk could this be a potential limitation of the library, at its current stage?

adam-urbanczyk, 2024-12-10:
If you don't like the generated orientation you can always use an aux spine to define one explicitly.

HamzaHassan9320, 2025-01-06:
> Same situation I think, sweeping a wire profile along a spline wave does not work properly (or as expected). Minimal example:
> 
> import cadquery as cq import numpy as np
> 
> support = cq.Workplane("XY").rect(50, 100).offset2D(5) nodes = [support.consolidateWires().val().positionAt(t) + cq.Vector(0, 0, 2+2_np.sin(20_np.pi*t)) for t in np.linspace(0,1,100)] sweepingPath = cq.Workplane("XY").polyline(nodes).close() # or .spline
> 
> ref0 = cq.Vector(nodes[0].y , nodes[0].z) points = [ ref0 + cq.Vector(5, 0), ref0 + cq.Vector(-5, 0), ref0 + cq.Vector(-5, -4) ]
> 
> profile = cq.Workplane("YZ", origin=(nodes[0].x,0,0)).polyline(points).consolidateWires() sweptPlane = profile.sweep(sweepingPath, isFrenet=False)
> 
> show_object(sweptPlane)
> 
> ![Screenshot 2024-12-10 070358](https://private-user-images.githubusercontent.com/87038052/394329625-a56afc73-2e61-4d71-bbd2-9f998fe2db43.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzEwMzIsIm5iZiI6MTczNjE3MDczMiwicGF0aCI6Ii84NzAzODA1Mi8zOTQzMjk2MjUtYTU2YWZjNzMtMmU2MS00ZDcxLWJiZDItOWY5OThmZTJkYjQzLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA2VDEzMzg1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTVlMTM0NTNlNzU3YjdiNWU3ZGQyMjU3OTU2NmM3MDkxYWVkZjVhOWZjYmVlZmZmNWI5MjcwZTM3NzVlOTNiMWMmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.be1Z8g2GXaFK4ZbOVpWbpbImHeU3FeUoIGW3i3a3yRI) ![Screenshot 2024-12-10 070439](https://private-user-images.githubusercontent.com/87038052/394329633-7ca4a80d-8f4b-4f61-8109-a478d1aec96b.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzEwMzIsIm5iZiI6MTczNjE3MDczMiwicGF0aCI6Ii84NzAzODA1Mi8zOTQzMjk2MzMtN2NhNGE4MGQtOGY0Yi00ZjYxLTgxMDktYTQ3OGQxYWVjOTZiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA2VDEzMzg1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPWQ3ZjRlMGM2MGM5OGEyYmMyZDFhZDYyMjE2Nzg3NjMwY2EyMTVjNzcwYTZiNjQ5OTEzN2VhYjE5MmEyOTg1ZGQmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.HKE65Py6t4RTeCwQNWwdyDkpkurNKBCGKRyDv0Qt2Y0) ![Screenshot 2024-12-10 070703](https://private-user-images.githubusercontent.com/87038052/394330273-b8051b51-adbe-48b6-b060-0c6ea251f769.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MzYxNzEwMzIsIm5iZiI6MTczNjE3MDczMiwicGF0aCI6Ii84NzAzODA1Mi8zOTQzMzAyNzMtYjgwNTFiNTEtYWRiZS00OGI2LWIwNjAtMGM2ZWEyNTFmNzY5LnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNTAxMDYlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjUwMTA2VDEzMzg1MlomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTM5MTBlMzFmYTM2NmI4NTg2ZGRkZTg3ZTJlYzAwNDk3MDUzYWJkOWZlNDA5ZDdlMzZkMmM4N2MyMWZhYjYwOTUmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0In0.TD-4ESnpq-dC-hb9FEEuuX4v11b_5UQez9IFCHrj8Jc) (nodes).close()

Did using an aux spine fix your issue @HTeyssedre? because I'm still facing similar problems.

adam-urbanczyk, 2025-02-03:
I suppose you are looking for this:
```python
from numpy import linspace, sin, cos, vstack, pi

from cadquery.func import *
from cadquery.vis import show

# params
t1 = 0  
t2 = 2
num_points = 100
radius = 50
offset_in_x = 5
circle_diameter = 4.0  
circle_radius = circle_diameter / 2

# paths
t = linspace(t1, t2, num_points)

x = radius * sin(2*pi*t)
y = radius * cos(2*pi*t)
z = 7 * sin(3.5 * 2*pi*t) + (14+circle_radius)*t

xa = 1.2*x
ya = 1.2*y
za = z

spine = spline(*vstack((x,y,z)).T.tolist())
aux_spine = spline(*vstack((xa,ya,za)).T.tolist())

# section
sect = (
    plane(circle_radius, 5*circle_radius)
    .moved(ry=90)
    .moved(spine.startPoint())
)

# final construction
res = sweep(sect, spine, aux_spine)

show(res)
```

[image]

HTeyssedre, 2025-02-13:
> Hello,
>
> Sorry, little time to reply to the previous mail. Did not find a solution
> by myself, tried to get some origin, normal, second guide and so on... But
> nothing came up as clean solution, I mean, could not make it work. I am
> focusing on another part of the capabilities right now and planning to look
> into it later. I will check the code that was just shared and give it
> another try this month.
>
> Thanks for the support!

Ok, so I gave it couple of tries, but I cannot make it work. Looks like a
calculation is made according to the center of the workplane, not relative
to the node position. If that is the case, you cannot catch it with a
radial screw sweep... here is some code, it's super slow, but just to show
the behavior:

def elevate_profile(wpObject, Vfunc, Npts):
    return [wpObject.consolidateWires().val().positionAt(t) + Vfunc(t) for
t in np.linspace(0,1,Npts)]

def expend(wpObject, t, orthVect, mag):
    return
mag*wpObject.consolidateWires().val().tangentAt(t).cross(orthVect)

def wave(t):
    return 20*cq.Vector(0, 0, 2*np.cos(2*np.pi*t)**2)

support = cq.Workplane("XY").rect(50, 100).offset2D(5)

nodes = elevate_profile(support, lambda u: wave(u), Npts = 100)
nodes_aux = elevate_profile(support, lambda u: wave(u) + expend(support, u,
cq.Vector(0,0,1), mag = 1), Npts = 100)

sweepingPath = cq.Workplane("XY").polyline(nodes).close()[0]
sweepingAux = cq.Workplane("XY").polyline(nodes_aux).close()[0]

show_object(sweepingPath)
show_object(sweepingAux)

ref0 = cq.Vector(nodes[0].y , nodes[0].z)
points = [ ref0 + cq.Vector(5, 0),
          ref0 + cq.Vector(5, 5),
          ref0 + cq.Vector(0, 5),
          ref0 + cq.Vector(0, 0)
        ]

# profile = cq.Workplane("YZ",
origin=(nodes[0].x,-2.5,-2.5)).polyline(points).close().consolidateWires()
# with a polyline profile
profile = cq.Workplane("YZ",
origin=(nodes[0].x,nodes[0].y,nodes[0].z)).circle(5) #  with a circle
show_object(profile)

# Sweep
sweptPlane = profile.sweep(sweepingPath, isFrenet=False,
auxSpine=sweepingAux, transition = 'round') #  ,
auxSpine=cq.Wire.makePolygon([nodes[0],nodes[0]+cq.Vector(0, 1, 0)])
show_object(sweptPlane.translate((0,0,30)))

Let me know if I did something wrong.

Best,
Hubert

HTeyssedre, 2025-02-13:
as a sanity check, if I turn of the wavy elevation (z=0) and change the
guides form polyline to spline, not what I expect to have, still some weird
deformation close to the fillets.

[image: image.png]