CadQuery/cadquery #1725: Prep 2.5

adam-urbanczyk, 2024-12-06:
- [ ] Merge outstanding PRs
- [ ] Final pass on free funcs
- [ ] Add arbitrary vtkActor rendering capabilities 
- [ ] Add non-blocking show (optional, not sure if easy)
- [ ] Restructure docs

adam-urbanczyk, 2024-12-06:
@jmwright @lorenzncode is there anything you'd like to add to the list?

huskier, 2024-12-08:
Prior to the release of V2.5.0, I recommend enhancing the show function in the vis.py file by adding a new parameter named "title", which defaults to "CQ viewer".

`win.SetWindowName("CQ viewer")
`

```
def show(
    *objs: Showable,
    scale: float = 0.2,
    alpha: float = 1,
    tolerance: float = 1e-3,
    edges: bool = False,
    **kwrags: Any,
):
    """
    Show CQ objects using VTK.
    """
    ......
    # create a VTK window
    win = _vtkRenderWindow(assy, tolerance=tolerance)

    win.SetWindowName("CQ viewer")
```

lorenzncode, 2024-12-08:
I have only #1713. As part of that I need to check whether any changes are required for the sphinx multimethod customization with sphinx version bump (and perhaps consider trying the `sphinx-immaterial` theme for docs as mentioned in #1697?).

closed 2025-01-01 by adam-urbanczyk