CadQuery/cadquery #1728: Cylinder surface incorrect Axis Direction

redsk, 2024-12-17:
Hi, I'm implementing some "drilling" for my project and therefore I need to detect holes in parts.

The orientation returned by `face.wrapped.TShape().Surface().Axis().Direction().Coord()` seems to be correct in one case of the cases (`c` below) and incorrect in others (`a` and `b`). 

However, for `c` the direction axis is parallel to `x` (correct) while for `a` and `b` it is parallel to `z` (wrong).

Why is this so?

```python
import cadquery as cq
from dataclasses import dataclass

@dataclass
class Hole():
    radius: float
    center: cq.cq.VectorLike
    direction: cq.cq.VectorLike

def find_holes(p: cq.Workplane, hole_size, tolerance=0.001):
    holes = []
    for f in p.faces():
        if (f.geomType() == 'CYLINDER'):
            surface = f.wrapped.TShape().Surface()
            radius = surface.Radius()
            if abs(radius - hole_size) < tolerance:
                direction = surface.Axis().Direction().Coord()
                center = f.CenterOfBoundBox()
                holes.append(Hole(radius, center, direction))
    return holes

def get_solid():
    return (
        cq.Workplane("YZ")
        .moveTo(0, 0)
        .lineTo(0, 10)
        .lineTo(10, 10)
        .lineTo(10, 0)
        .close()
        .extrude(2, both=True)
    )

a = (
    get_solid()
    .workplane(2)
    .pushPoints([(5, 5)])
    .hole(2)
)

b = (
    get_solid()
    .pushPoints([(5, 5)])
    .cylinder(4, 1, combine='cut')
)

c = (
    get_solid()
    .pushPoints([(5, 5)])
    .circle(1)
    .cutThruAll()
)

def list_holes(pl):
    holes = find_holes(pl, 1)
    for h in holes:
        print(h)
    
print("holes in a:")
list_holes(a)

print("\nholes in b:")
list_holes(b)

print("\nholes in c:")
list_holes(c)
```

Output:
```
holes in a:
Hole(radius=1.0, center=Vector: (0.0, 5.0, 5.0), direction=(0.0, 0.0, -1.0))

holes in b:
Hole(radius=1.0, center=Vector: (0.0, 5.0, 5.0), direction=(0.0, 0.0, 1.0))

holes in c:
Hole(radius=1.0, center=Vector: (0.0, 5.0, 5.0), direction=(-1.0, 0.0, 0.0))
```

lorenzncode, 2024-12-17:
> Why is this so?

The shapes also have Location.  It looks like this low level code did not consider it.

Try without the low level OCCT.

```py
import cadquery as cq

a = (
    cq.Workplane("YZ")
    .moveTo(0, 0)
    .lineTo(0, 10)
    .lineTo(10, 10)
    .lineTo(10, 0)
    .close()
    .extrude(2, both=True)
)

a = a.workplane(2).pushPoints([(5, 5)]).hole(2)

center_cyl = a.faces("%CYLINDER").val().Center()
edge1 = a.faces("%CYLINDER").edges("%CIRCLE").val()
radius_cyl = edge1.radius()
normal_cyl = edge1.normal()
```

redsk, 2024-12-17:
> The shapes also have Location. It looks like this low level code did not consider it.
> 
> Try without the low level OCCT.

Mine was just a minimal example, but how do I find the normal in this case?

```python
rod = (
    cq.Workplane("XY")
    .cylinder(30, 10)
    .transformed(rotate=(90, 0, 0))
    .pushPoints([(0, 0)])
    .cylinder(30, 5, combine='cut')
)
```

Here the edges are not `CIRCLE`, they are `BSPLINE`, and they don't have the `normal()` method.

adam-urbanczyk, 2024-12-17:
This is not an issue anymore, converting to a discussion.