CadQuery/cadquery #1740: TypeError: Sketch.moved() got an unexpected keyword argument 'z'

WayneRiesterer, 2025-01-02:
## To Reproduce
Attempt to render the following example from the documentation section _Lofting between two sketches_ in CQ Editor:

```
from cadquery import Workplane, Sketch, Vector, Location

s1 = Sketch().trapezoid(3, 1, 110).vertices().fillet(0.2)

s2 = Sketch().rect(2, 1).vertices().fillet(0.2)

result = Workplane().placeSketch(s1, s2.moved(z=3)).loft()
```

[image]

## Backtrace
```
TypeError: Sketch.moved() got an unexpected keyword argument 'z'
File: <cq_editor-string>
Line: 7
Code: result = Workplane().placeSketch(s1, s2.moved(z=3)).loft()
```

## This works
```
from cadquery import Workplane, Sketch, Vector, Location

s1 = Sketch().trapezoid(3, 1, 110).vertices().fillet(0.2)

s2 = Sketch().rect(2, 1).vertices().fillet(0.2)

# replace z=3 with Vector(0,0,3)
result = Workplane().placeSketch(s1, s2.moved(Vector(0,0,3))).loft()
```

[image]

## Environment
OS: Windows 11 Pro
Using: CQ-Editor Version 0.3.0dev

adam-urbanczyk, 2025-01-02:
I cannot reproduce this on master.

WayneRiesterer, 2025-01-02:
Does this suggest it won't be an issue if I get the code from Master?

adam-urbanczyk, 2025-01-02:
Yes

WayneRiesterer, 2025-01-02:
Thanks Adam.