CadQuery/cadquery #1742: Dependencies not installed with 2.5

kmarchais, 2025-01-03:
Hello, since release 2.5, the dependencies are not installed automatically like it was before.

But if I install myself the dependencies of cadquery, it works fine:
```python
reqs = [
    "cadquery-ocp",
    "ezdxf",
    "multimethod",
    "nlop",
    "typish",
    "casadi",
    "path",
]
```

jmwright, 2025-01-03:
@kmarchais Are you using an Anaconda environment?

kmarchais, 2025-01-03:
No, I use `pip install cadquery`.

adam-urbanczyk, 2025-01-03:
But are you by any chance calling it within a conda env?

kmarchais, 2025-01-03:
No I am not in a conda env but after checking I had a env var CONDA_PREFIX set.
After removing it, the issue persists though

jmwright, 2025-01-03:
Actually, I am able to reproduce this. When I do `pip install -e .` locally without conda initted, all the dependencies install. When I follow the same process with `pip install cadquery`, the dependencies are not installed.

```
(cq-dev) jwright@lmde6:~/repos/cadquery$ pip install cadquery
Collecting cadquery
  Downloading cadquery-2.5.1-py3-none-any.whl (165 kB)
     ━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━━ 165.6/165.6 kB 2.4 MB/s eta 0:00:00
Installing collected packages: cadquery
Successfully installed cadquery-2.5.1
```

Is all I see when installing from PyPI. When I check the environment, there is no mention of conda.

adam-urbanczyk, 2025-01-03:
Then I did something wrong when building the package.

jmwright, 2025-01-03:
The dependenices did not make it into the METADATA file when the package was uploaded to PyPI.

[image]

I believe I had this happen once on a previous release. I believe the issue happened during the build phase (i.e. the conda environment variable was present in the environment during the build).

adam-urbanczyk, 2025-01-03:
yes, see my comment above :)

jmwright, 2025-01-03:
Item 5 under [Publish to PyPI](https://github.com/CadQuery/cadquery/wiki/Developer-Doc:-Creating-a-New-Release#publish-to-pypi-reference) in the wiki I think describes this problem.

jmwright, 2025-01-03:
@adam-urbanczyk Sorry, we are posting at the same time. I see your comments above now.

jmwright, 2025-01-03:
@kmarchais This has been fixed now. Please use the `--upgrade` flag with pip if you want to update cadquery in your existing environment.

https://pypi.org/manage/project/cadquery/release/2.5.2/

closed 2025-01-03 by jmwright