CadQuery/cadquery #1747: import issue bug  "from OCP.OCP import *"

wangbochao1, 2025-01-09:
One month ago, I installed cadquery (cadquery==2.5.0. dev0, cadquery ocp==7.7.2) in the following way and installed it normally. The code can run normally;
Today I also installed it in the same way (cadquery==2.4.0, cadquery ocp==7.7.2.2b2), and there were no problems with the installation, but there was an error when executing the code. Please refer to the bottom for details

conda create -n cq_test_1 python=3.11
python3 -m pip install --upgrade pip
pip install cadquery

Traceback (most recent call last):
  File "<frozen importlib._bootstrap>", line 1176, in _find_and_load
  File "<frozen importlib._bootstrap>", line 1147, in _find_and_load_unlocked
  File "<frozen importlib._bootstrap>", line 690, in _load_unlocked
  File "<frozen importlib._bootstrap_external>", line 940, in exec_module
  File "<frozen importlib._bootstrap>", line 241, in _call_with_frames_removed
  File "/Users/wangbochao/miniconda3/envs/cq_test_1/lib/python3.11/site-packages/cadquery/__init__.py", line 10, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "/Users/wangbochao/miniconda3/envs/cq_test_1/lib/python3.11/site-packages/cadquery/occ_impl/geom.py", line 5, in <module>
    from OCP.gp import (
  File "/Users/wangbochao/miniconda3/envs/cq_test_1/lib/python3.11/site-packages/OCP/__init__.py", line 14, in <module>
    from OCP.OCP import *
ImportError: dlopen(/Users/wangbochao/miniconda3/envs/cq_test_1/lib/python3.11/site-packages/OCP/OCP.cpython-311-darwin.so, 0x0002): Library not loaded: @rpath/libvtkWrappingPythonCore3.11-9.2.dylib
  Referenced from: <E1C9E680-B854-386F-BFE3-ECC5FF8F3044> /Users/wangbochao/miniconda3/envs/cq_test_1/lib/python3.11/site-packages/OCP/OCP.cpython-311-darwin.so
  Reason: tried: '/Users/wangbochao/miniconda3/envs/cq_test_1/lib/python3.11/site-packages/OCP/../vtkmodules/.dylibs/libvtkWrappingPythonCore3.11-9.2.dylib' (no such file), '/Users/wangbochao/miniconda3/envs/cq_test_1/lib/python3.11/site-packages/OCP/../vtkmodules/.dylibs/libvtkWrappingPythonCore3.11-9.2.dylib' (no such file), '/Users/wangbochao/miniconda3/envs/cq_test_1/bin/../lib/libvtkWrappingPythonCore3.11-9.2.dylib' (no such file), '/Users/wangbochao/miniconda3/envs/cq_test_1/bin/../lib/libvtkWrappingPythonCore3.11-9.2.dylib' (no such file), '/usr/local/lib/libvtkWrappingPythonCore3.11-9.2.dylib' (no such file), '/usr/lib/libvtkWrappingPythonCore3.11-9.2.dylib' (no such file, not in dyld cache)
python-BaseException

jmwright, 2025-01-09:
What code are you trying to run and what OS and architecture are you running on?

wangbochao1, 2025-01-09:
Here is my code.

outer_diameter=29.8
inner_diameter=18.5
height=10

middle_diameter=24.1
inner_height=4.3

fillet_radius_1=1
fillet_radius_2=1.5

second_ring_outer_diameter = outer_diameter
second_ring_inner_diameter=24.1
second_ring_height=1.5

groove=27.2
groove_length=6

import cadquery as cq
from cadquery import exporters
import math

ring = (
    cq.Workplane("XY")
    .circle(outer_diameter / 2)
    .circle(inner_diameter / 2)
    .extrude(height)
)

inner_cylinder = cq.Workplane("XY").circle(middle_diameter / 2).extrude(inner_height)

inner_cylinder = inner_cylinder.faces(">Z").fillet(fillet_radius_2)

ring = ring.cut(inner_cylinder)

second_ring = (
    cq.Workplane("XY", origin=(0,0,height))
    .circle(second_ring_outer_diameter / 2)  # 外圆
    .circle(second_ring_inner_diameter / 2)  # 内圆
    .extrude(-second_ring_height) # 向下生成环形
)

bottom_inner_edges = (
    second_ring.edges("<Z")  # 选择底部的边
    .filter(lambda e: e.geomType() == "CIRCLE" and abs(e.radius() - second_ring_inner_diameter / 2) <= 0 )  # 过滤出内径边
)

second_ring_fillet = bottom_inner_edges.fillet(fillet_radius_1)
result = ring.cut(second_ring_fillet)

cut1 = (
    cq.Workplane("XY", origin=(outer_diameter/2,0,0)) # 设置右边为起始点
    .rect( (outer_diameter-groove), groove_length )
    .extrude(height)
)

cut2 = (
    cq.Workplane("XY", origin=(-outer_diameter/2,0,0)) # 设置右边为起始点
    .rect( (outer_diameter-groove), groove_length )
    .extrude(height)
)

result = result.cut(cut1).cut(cut2)

exporters.export(result, './result.step')

adam-urbanczyk, 2025-01-09:
I tried locally on win with pip (no mac at hand) and I have no issues.

lorenzncode, 2025-01-10:
> Now I have changed the version number to this: cadquery==2.4.0, cadquery-ocp==7.7.2 But it still reports an error Workplane' object has no attribute 'filter'. Did you mean: '_filter'?

This error is expected with 2.4.  `filter` is new in 2.5.

wangbochao1, 2025-01-10:
Thank you all for your help

closed 2025-01-10 by wangbochao1