CadQuery/cadquery #1754: sketch + offset + cut gives Standard_Failure: BRep_API: command not done

louisabraham, 2025-01-26:
```
s = cq.Sketch().rect(10, 2).moved(5, 0, 0).circle(2).clean()
r = (
    cq.Workplane("XY")
    .box(unit_length, unit_length, unit_length)
    .faces(">X")
    .workplane()
    .placeSketch(s)
    .cutThruAll()
)
```

works and produces this

[image]

```
s = cq.Sketch().rect(10, 2).moved(5, 0, 0).circle(2).clean().wires().offset(1)
r = (
    cq.Workplane("XY")
    .box(unit_length, unit_length, unit_length)
    .faces(">X")
    .workplane()
    .placeSketch(s)
    .cutThruAll()
)
```

gives

```
Standard_Failure                          Traceback (most recent call last)
Cell In[58], [line 9](vscode-notebook-cell:?execution_count=58&line=9)
      [1](vscode-notebook-cell:?execution_count=58&line=1) # %%
      [2](vscode-notebook-cell:?execution_count=58&line=2) s = cq.Sketch().rect(10, 2).moved(5, 0, 0).circle(2).clean().wires().offset(1)
      [3](vscode-notebook-cell:?execution_count=58&line=3) r = (
      [4](vscode-notebook-cell:?execution_count=58&line=4)     cq.Workplane("XY")
      [5](vscode-notebook-cell:?execution_count=58&line=5)     .box(unit_length, unit_length, unit_length)
      [6](vscode-notebook-cell:?execution_count=58&line=6)     .faces(">X")
      [7](vscode-notebook-cell:?execution_count=58&line=7)     .workplane()
      [8](vscode-notebook-cell:?execution_count=58&line=8)     .placeSketch(s)
----> [9](vscode-notebook-cell:?execution_count=58&line=9)     .cutThruAll()
     [10](vscode-notebook-cell:?execution_count=58&line=10) )
     [11](vscode-notebook-cell:?execution_count=58&line=11) reset_show()
     [12](vscode-notebook-cell:?execution_count=58&line=12) show_object(r, options={"color": "red"})

File ~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3661, in Workplane.cutThruAll(self, clean, taper)
   [3645](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3645) """
   [3646](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3646) Use all un-extruded wires in the parent chain to create a prismatic cut from existing solid.
   [3647](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3647) Cuts through all material in both normal directions of workplane.
   (...)
   [3657](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3657) see :meth:`cutBlind` to cut material to a limited depth
   [3658](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3658) """
   [3659](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3659) solidRef = self.findSolid()
-> [3661](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3661) s = solidRef.dprism(
   [3662](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3662)     None, self._getFaces(), thruAll=True, additive=False, taper=-taper
   [3663](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3663) )
   [3665](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3665) if clean:
   [3666](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/cq.py:3666)     s = s.clean()

File ~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/multimethod/__init__.py:375, in multimethod.__call__(self, *args, **kwargs)
    [373](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/multimethod/__init__.py:373) func = self.dispatch(*args)
    [374](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/multimethod/__init__.py:374) try:
--> [375](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/multimethod/__init__.py:375)     return func(*args, **kwargs)
    [376](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/multimethod/__init__.py:376) except TypeError as ex:
    [377](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/multimethod/__init__.py:377)     raise DispatchError(f"Function {func.__code__}") from ex

File ~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py:3479, in Mixin3D.dprism(self, basis, faces, depth, taper, upToFace, thruAll, additive)
   [3476](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py:3476)     else:
   [3477](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py:3477)         feat.Perform(depth)
-> [3479](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py:3479)     shape = feat.Shape()
   [3481](https://file+.vscode-resource.vscode-cdn.net/Users/louis/code/scaccomatto/~/.pyenv/versions/3.11.10/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py:3481) return self.__class__(shape)

Standard_Failure: BRep_API: command not done
```

adam-urbanczyk, 2025-01-27:
If you visualize `s` it'll be clear. Probably you want `s = cq.Sketch().rect(10, 2).moved(5, 0, 0).circle(2).clean().wires().offset(1, mode='r')`

louisabraham, 2025-01-29:
Indeed, it works! However I don't see any difference when visualizing. I'm using OCP CAD Viewer.

adam-urbanczyk, 2025-01-30:
I do see a clear difference:

[image]

louisabraham, 2025-01-30:
I mean between the two versions of `s`!
Of course the two working full codes produce different holes. But I understood you hinted that the `"r"` option would modify the sketches in a visible way.

adam-urbanczyk, 2025-01-30:
Then visualize both using `show` from `cadquery.vis` (see [here](https://cadquery.readthedocs.io/en/latest/vis.html#pure-python)). If you are still confused, paste a screenshot.

louisabraham, 2025-01-30:
oh thanks! it is clear now! I was using OCP CAD Viewer.

for reference here is the output of `cadquery.vis.show`

[image]

[image]

closed 2025-01-30 by louisabraham