CadQuery/cadquery #1755: Cutting a solid constructed from edges not working as expected

snowbldr, 2025-01-28:
I'm trying to construct arbitrary polyhedrons based on vertex and face definitions, then cut those solids to place text on their faces (making dice). 

The issue I'm having is that after I construct the solid and cut the face, the face gets deleted. 

[image]

If I switch to constructing a box instead of using Solid.makeSolid, the cut works fine

[image]

Here's the code I'm using:
```python
from cadquery import Workplane, Plane, Vector, Face, Solid, Shell, Wire, Edge
from cadquery.vis import show_object

plane = Workplane()
t = 1 / (3 ** 0.5)
vertexes = (
    (t, t, t), (-t, t, t), (-t, -t, t), (t, -t, t),
    (t, t, -t), (-t, t, -t), (-t, -t, -t), (t, -t, -t)
)

faces = (
    (0, 1, 2, 3), (0, 1, 5, 4), (0, 3, 7, 4),
    (6, 7, 4, 5), (6, 7, 3, 2), (6, 5, 1, 2),
)

w = Workplane(Solid.makeSolid(Shell.makeShell(
    Face.makeFromWires(Wire.assembleEdges([
        Edge.makeLine(
            points[i],
            points[i + 1 if i < len(points) else 0])
        for i in range(len(points) - 1)
    ])) for points in [[vertexes[i] for i in face] for face in faces]
)))

# using this below works fine?
# w = Workplane().box(1,1,1)

i=0
f: Face = w.faces().vals()[i]
world_up = Vector(0, 0, 1).normalized()
origin = f.Center()
normal = f.normalAt()
x_dir = world_up.cross(normal)

symbol = (
    Workplane(Plane(origin, x_dir, normal), origin=origin)
    .text(
        '1',
        font='Arial',
        fontsize=1,
        distance=-.25
    )
)

show_object(plane.add(w.cut(symbol)))

```

Any help with this would be greatly appreciated as I don't understand what's going on here.

lorenzncode, 2025-01-30:
>  #using this below works fine?
>  #w = Workplane().box(1,1,1)

Are you trying to make a box with the code above the comment? It only has 10 edges.

`show_object(w.edges())`:
[image]

`w.val()` is not valid.  You can check the shape with `isValid` method or use `check` with `from cadquery.func import check`. 

Try to get a valid shape first then proceed with the cut.

snowbldr, 2025-01-30:
Ah, that was it! I was iterating my edges incorrectly.

Thanks a ton for your help and pointing out those handy debugging tricks.

Here's my fixed code for posterity (and an example of creating a 6 sided die from a set of vertexes and faces):
```python
from cadquery import Workplane, Plane, Vector, Face, Solid, Shell, Wire, Edge
from cadquery.vis import show_object

t = 1 / (3 ** 0.5)
vertexes = (
    (t, t, t), (-t, t, t), (-t, -t, t), (t, -t, t),
    (t, t, -t), (-t, t, -t), (-t, -t, -t), (t, -t, -t)
)

faces = (
    (0, 1, 2, 3), (0, 1, 5, 4), (0, 3, 7, 4),
    (6, 7, 4, 5), (6, 7, 3, 2), (6, 5, 1, 2),
)

face_points = [[vertexes[i] for i in face] for face in faces]
faces = []
for points in face_points:
    faces.append(
        Face.makeFromWires(Wire.assembleEdges(
            Edge.makeLine(
                points[i],
                points[i + 1 if i < len(points) - 1 else 0]
            ) for i in range(len(points))))
    )

w = Workplane(Solid.makeSolid(Shell.makeShell(faces)))

for i, face in enumerate(w.faces().vals()):
    world_up = Vector(0, 0, 1).normalized()
    origin = face.Center()
    normal = face.normalAt()
    x_dir = world_up.cross(normal)
    if x_dir.Length == 0:
        x_dir = (1,0,0)

    symbol = (
        Workplane(Plane(origin, x_dir, normal), origin=origin)
        .text(
            str(i+1),
            font='Arial',
            fontsize=1,
            distance=-.025
        )
    )

    w = w.cut(symbol)

show_object(w)
```

closed 2025-01-30 by snowbldr