CadQuery/cadquery #1756: TopoDS_Face HashCode Method is Missing

jmwright, 2025-01-28:
## To Reproduce

1. Install CadQuery master from conda: `mamba install -c conda-forge -c cadquery cadquery=master`
2. Run the following code

```python
import cadquery as cq

res = cq.Workplane().box(10, 10, 10)

res = res.faces(">Z").workplane().circle(2.0).cutThruAll()
```

## Backtrace

```
Traceback (most recent call last):
  File "<stdin>", line 1, in <module>
  File "/home/jwright/.local/lib/python3.11/site-packages/cadquery/cq.py", line 917, in faces
    return self._selectObjects("Faces", selector, tag)
           ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jwright/.local/lib/python3.11/site-packages/cadquery/cq.py", line 832, in _selectObjects
    toReturn = cq_obj._collectProperty(objType)
               ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^
  File "/home/jwright/.local/lib/python3.11/site-packages/cadquery/cq.py", line 252, in _collectProperty
    rv[k] = None
    ~~^^^
  File "/home/jwright/.local/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py", line 1208, in __hash__
    return self.hashCode()
           ^^^^^^^^^^^^^^^
  File "/home/jwright/.local/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py", line 599, in hashCode
    return self.wrapped.HashCode(HASH_CODE_MAX)
           ^^^^^^^^^^^^^^^^^^^^^
AttributeError: 'OCP.OCP.TopoDS.TopoDS_Face' object has no attribute 'HashCode'
```

## Environment

OS: Debian 12, 64-bit AMD processor

Using: The terminal or CQ-editor with the latest CadQuery installed.

Was CadQuery installed using Conda?: `Yes`

Output of `conda list` from your active Conda environment:
```
# packages in environment at /home/jwright/mambaforge/envs/cq:
#
# Name                    Version                   Build  Channel
_libgcc_mutex             0.1                 conda_forge    conda-forge
_openmp_mutex             4.5                       2_gnu    conda-forge
aiohappyeyeballs          2.4.4              pyhd8ed1ab_1    conda-forge
aiohttp                   3.11.11         py311h2dc5d0c_0    conda-forge
aiosignal                 1.3.2              pyhd8ed1ab_0    conda-forge
alsa-lib                  1.2.13               hb9d3cd8_0    conda-forge
ampl-asl                  1.0.0                h5888daf_2    conda-forge
aom                       3.9.1                hac33072_0    conda-forge
asttokens                 3.0.0              pyhd8ed1ab_1    conda-forge
attrs                     25.1.0             pyh71513ae_0    conda-forge
blosc                     1.21.6               he440d0b_1    conda-forge
brotli                    1.1.0                hb9d3cd8_2    conda-forge
brotli-bin                1.1.0                hb9d3cd8_2    conda-forge
bzip2                     1.0.8                h4bc722e_7    conda-forge
c-ares                    1.34.4               hb9d3cd8_0    conda-forge
ca-certificates           2024.12.14           hbcca054_0    conda-forge
cadquery                  master          v2.5.2_4_gf13f607    cadquery
cairo                     1.18.2               h3394656_1    conda-forge
casadi                    3.6.7           py311h5f149b6_2    conda-forge
cyrus-sasl                2.1.27               h54b06d7_7    conda-forge
dav1d                     1.2.1                hd590300_0    conda-forge
[... 210 lines cut ...]
xcb-util-wm               0.4.2                hb711507_0    conda-forge
xkeyboard-config          2.43                 hb9d3cd8_0    conda-forge
xorg-libice               1.1.2                hb9d3cd8_0    conda-forge
xorg-libsm                1.2.5                he73a12e_0    conda-forge
xorg-libx11               1.8.10               h4f16b4b_1    conda-forge
xorg-libxau               1.0.12               hb9d3cd8_0    conda-forge
xorg-libxcomposite        0.4.6                hb9d3cd8_2    conda-forge
xorg-libxcursor           1.2.3                hb9d3cd8_0    conda-forge
xorg-libxdamage           1.1.6                hb9d3cd8_0    conda-forge
xorg-libxdmcp             1.1.5                hb9d3cd8_0    conda-forge
xorg-libxext              1.3.6                hb9d3cd8_0    conda-forge
xorg-libxfixes            6.0.1                hb9d3cd8_0    conda-forge
xorg-libxi                1.8.2                hb9d3cd8_0    conda-forge
xorg-libxrandr            1.5.4                hb9d3cd8_0    conda-forge
xorg-libxrender           0.9.12               hb9d3cd8_0    conda-forge
xorg-libxt                1.3.1                hb9d3cd8_0    conda-forge
xorg-libxtst              1.2.5                hb9d3cd8_3    conda-forge
xorg-libxxf86vm           1.1.6                hb9d3cd8_0    conda-forge
xz                        5.6.3                hbcc6ac9_1    conda-forge
xz-gpl-tools              5.6.3                hbcc6ac9_1    conda-forge
xz-tools                  5.6.3                hb9d3cd8_1    conda-forge
yarl                      1.18.3          py311h2dc5d0c_1    conda-forge
zlib                      1.3.1                hb9d3cd8_2    conda-forge
zstd                      1.5.6                ha6fb4c9_0    conda-forge
```

adam-urbanczyk, 2025-01-28:
This method does not exist, see
https://github.com/CadQuery/cadquery/blob/f13f60735ff0593524097eb6ed47f97ef1ee722a/cadquery/occ_impl/shapes.py#L598

Something wrong in your env? AFAICT last upload of the master package (v2.5.2_4_gf13f607) went fine.

jmwright, 2025-01-28:
@adam-urbanczyk I've done two different fresh environments and had the same issue. Maybe we can wait to see if anyone else has the issue. Maybe it's my mamba version or something.

adam-urbanczyk, 2025-01-28:
Could you check what do you actually see in your sources? I.e. is it really pulled from the conda package or  from smth else? `/home/jwright/.local/lib/python3.11/site-packages/cadquery/occ_impl/shapes.py` does not look like a conda env to me, but who knows.

jmwright, 2025-01-28:
@adam-urbanczyk The `mamba` command is somehow broken on my system. I have not updated, but if I create the environment and do installs with `mamba`, my local Python install is used. If I use `conda`, everything works as expected with the conda environment being used. I am going to close this issue and assume that it's just that my system is broken.