CadQuery/cadquery #1757: ValueError: Null TopoDS_Shape object with offset of circle

louisabraham, 2025-01-29:
```
ellipsis_big = 1
ellipsis_small = 1
s = cq.Sketch().ellipse(ellipsis_big, ellipsis_small).wires().offset(-0.1, mode="s")
show(s)
```

Any value of ellipsis_small different than 1 will not cause a bug. Using `circle` or `ellipsis_small = ellipsis_big` causes `ValueError: Null TopoDS_Shape object`. Currently I use `0.999` to alleviate the issue.

gart-samjon701823, 2025-01-29:
Hello, we tried to solve the issue.

This is what we did:

Update the ellipse method in the Sketch class to handle the case where both axes are equal, which would effectively create a circle. This fixes the issue where using equal values for ellipsis_big and ellipsis_small causes a ValueError.

You can review changes in this commit: https://github.com/gart-samjon701823/CadQuery-cadquery-1757/commit/495c8e472fedb67955dc7fd12e41461afb56e246.

> [!CAUTION]
> Disclaimer: The concept of solution was created by AI and you should never copy paste this code before you check the correctness of generated code. Solution might not be complete, you should use this code as an inspiration only.

---

Latta AI seeks to solve problems in open source projects as part of its mission to support developers around the world. Learn more about our mission at https://latta.ai/ourmission . If you no longer want Latta AI to attempt solving issues on your repository, you can block this account.

adam-urbanczyk, 2025-02-03:
Confirmed, FWIW a workaround:
```python
import cadquery as cq

from cadquery.func import wire, ellipse
from math import pi

eps = 1e-6

s = cq.Sketch().face(wire(ellipse(1,1).trim(0,2*pi-eps)).wires().offset(-0.1, mode='s')
```