CadQuery/cadquery #1758: import issue

Vishwavijeth, 2025-01-30:
Traceback (most recent call last):
  File "F:\Prep\Step FIle Processing\step.py", line 1, in <module>
    import cadquery as cq
  File "F:\Prep\myvenv\Lib\site-packages\cadquery\__init__.py", line 10, in <module>
    from .occ_impl.geom import Plane, BoundBox, Vector, Matrix, Location
  File "F:\Prep\myvenv\Lib\site-packages\cadquery\occ_impl\geom.py", line 5, in <module>
    from OCP.gp import (
    ...<11 lines>...
    )
ModuleNotFoundError: No module named 'OCP'

jmwright, 2025-01-30:
@Vishwavijeth If you had conda active at all, or any of [these variables](https://github.com/CadQuery/cadquery/blob/master/setup.py#L24) in the environment when you installed via pip, cadquery-ocp and the other dependencies probably did not get installed. You can do a `pip list` and look for `cadquery-ocp` and the other dependencies listed [here](https://github.com/CadQuery/cadquery/blob/master/setup.py#L28).

Vishwavijeth, 2025-01-30:
Thank you @jmwright

closed 2025-01-30 by Vishwavijeth