CadQuery/cadquery #1763: Could the Free Function API and the Workplane API with the same function names be designed to have consistent parameter interfaces as much as possible?

huskier, 2025-02-06:
Could the Free Function API and the Workplane API with the same function names be designed to have consistent parameter interfaces as much as possible? Here is a table of inconsistencies between the Free Function API and the Workplane API with the same function names:
|Items|Free Function API|CadQuery Workplane API|
| - | - | - |
|rect|def rect(w: float, h: float) -> Shape:|<p>def rect(</p><p>`    `self: T,</p><p>`    `xLen: float,</p><p>`    `yLen: float,</p><p>`    `centered: Union[bool, Tuple[bool, bool]] = True,</p><p>`    `forConstruction: bool = False,</p><p>) -> T:	</p>|
|circle|def circle(r: float) -> Shape:|def circle(self: T, radius: float, forConstruction: bool = False) -> T:|
|ellipse|def ellipse(r1: float, r2: float) -> Shape:|<p>def ellipse(</p><p>`    `self: T,</p><p>`    `x\_radius: float,</p><p>`    `y\_radius: float,</p><p>`    `rotation\_angle: float = 0.0,</p><p>`    `forConstruction: bool = False,</p><p>) -> T:</p>|
|box|def box(w: float, l: float, h: float) -> Shape:|<p>def box(</p><p>`    `self: T,</p><p>`    `length: float,</p><p>`    `width: float,</p><p>`    `height: float,</p><p>`    `centered: Union[bool, Tuple[bool, bool, bool]] = True,</p><p>`    `combine: CombineMode = True,</p><p>`    `clean: bool = True,</p><p>) -> T:	</p>|
|cylinder|def cylinder(d: float, h: float) -> Shape:|<p>def cylinder(</p><p>`    `self: T,</p><p>`    `height: float,</p><p>`    `radius: float,</p><p>`    `direct: Union[Tuple[float, float, float], Vector] = Vector(0, 0, 1),</p><p>`    `angle: float = 360,</p><p>`    `centered: Union[bool, Tuple[bool, bool, bool]] = True,</p><p>`    `combine: CombineMode = True,</p><p>`    `clean: bool = True,</p><p>) -> T:</p>|
|sphere|def sphere(d: float) -> Shape:|<p>def sphere(</p><p>`    `self: T,</p><p>`    `radius: float,</p><p>`    `direct: VectorLike = (0, 0, 1),</p><p>`    `angle1: float = -90,</p><p>`    `angle2: float = 90,</p><p>`    `angle3: float = 360,</p><p>`    `centered: Union[bool, Tuple[bool, bool, bool]] = True,</p><p>`    `combine: CombineMode = True,</p><p>`    `clean: bool = True,</p><p>) -> T:	</p>|

adam-urbanczyk, 2025-02-06:
Fair request, there is also `cq.Sketch` to match. For clarity: are you referring to parameter order and semantics (e.g. radius vs diameter) or names too?

huskier, 2025-02-06:
All of them, I think:
1) the parameters order;
2) the semantics, by radius and diameter, I think radius is preferred.
3) the parameter name, for example, "width" is preferred to "w" for width.

We do not need to think about the parameters' thing when we write code. It is a natural thing for the parameters.

huskier, 2025-02-06:
Is it possible to keep the centered parameter for box, cylinder and sphere free functions? This is also related to consistency with the the same name methods in Workplane class.

adam-urbanczyk, 2025-02-06:
For now the design is to not have those and keep the api simple. In the future let's see.