CadQuery/cadquery #1764: ValueError: offset() 'Invalid mode: r'  in sketch example

gitterman, 2025-02-08:
I tried the examples on https://cadquery.readthedocs.io/en/latest/sketch.html#sketches-spanning-multiple-elements and get the error 'Invalid mode: r' when I specify this mode.
The offset() page mentions  "mode (Literal['a', 's', 'i', 'c', 'r'])" but there is no explanation on these and mode 'r' doesn't seem to exist.

adam-urbanczyk, 2025-02-08:
It does work:
[image]

You are likely using older version of CQ. Could you follow the issue template and add the output of `conda list` or similar.

gitterman, 2025-02-09:
Sorry, I did not think of this.
I am using CQ Editor 0.3.0dev and below is the output of 'conda list'.

```
# packages in environment at C:\ProgramData\miniconda3:
#
# Name                    Version                   Build  Channel
archspec                  0.2.1              pyhd3eb1b0_0  
boltons                   23.0.0          py312haa95532_0  
brotli-python             1.0.9           py312hd77b12b_7  
bzip2                     1.0.8                he774522_0  
ca-certificates           2023.12.12           haa95532_0  
certifi                   2024.2.2        py312haa95532_0  
cffi                      1.16.0          py312h2bbff1b_0  
charset-normalizer        2.0.4              pyhd3eb1b0_0  
colorama                  0.4.6           py312haa95532_0  
conda                     24.1.2          py312haa95532_0  
conda-content-trust       0.2.0           py312haa95532_0  
conda-libmamba-solver     23.12.0            pyhd3eb1b0_1  
conda-package-handling    2.2.0           py312haa95532_0  
conda-package-streaming   0.9.0           py312haa95532_0  
console_shortcut_miniconda 0.1.1                haa95532_1  
cryptography              41.0.7          py312h89fc84f_0  
distro                    1.8.0           py312haa95532_0  
expat                     2.5.0                hd77b12b_0  
fmt                       9.1.0                h6d14046_0  
idna                      3.4             py312haa95532_0  
jsonpatch                 1.32               pyhd3eb1b0_0  
jsonpointer               2.1                pyhd3eb1b0_0  
libarchive                3.6.2                hb62f4d4_2  
libcurl                   8.5.0                h86230a5_0  
libffi                    3.4.4                hd77b12b_0  
libiconv                  1.16                 h2bbff1b_2  
libmamba                  1.5.3                hcd6fe79_0  
libmambapy                1.5.3           py312h77c03ed_0  
libsolv                   0.7.24               h23ce68f_0  
libssh2                   1.10.0               he2ea4bf_2  
libxml2                   2.10.4               h0ad7f3c_1  
lz4-c                     1.9.4                h2bbff1b_0  
menuinst                  2.0.2           py312hd77b12b_0  
openssl                   3.0.13               h2bbff1b_0  
packaging                 23.1            py312haa95532_0  
pcre2                     10.42                h0ff8eda_0  
pip                       23.3.1          py312haa95532_0  
platformdirs              3.10.0          py312haa95532_0  
pluggy                    1.0.0           py312haa95532_1  
powershell_shortcut_miniconda 0.0.1                haa95532_1  
pybind11-abi              4                    hd3eb1b0_1  
pycosat                   0.6.6           py312h2bbff1b_0  
pycparser                 2.21               pyhd3eb1b0_0  
pysocks                   1.7.1           py312haa95532_0  
python                    3.12.1               h1d929f7_0  
reproc                    14.2.4               hd77b12b_1  
reproc-cpp                14.2.4               hd77b12b_1  
requests                  2.31.0          py312haa95532_1  
ruamel.yaml               0.17.21         py312h2bbff1b_0  
setuptools                68.2.2          py312haa95532_0  
sqlite                    3.41.2               h2bbff1b_0  
tk                        8.6.12               h2bbff1b_0  
tqdm                      4.65.0          py312hfc267ef_0  
truststore                0.8.0           py312haa95532_0  
tzdata                    2023d                h04d1e81_0  
urllib3                   2.1.0           py312haa95532_1  
vc                        14.2                 h21ff451_1  
vs2015_runtime            14.27.29016          h5e58377_2  
wheel                     0.41.2          py312haa95532_0  
win_inet_pton             1.1.0           py312haa95532_0  
xz                        5.4.5                h8cc25b3_0  
yaml-cpp                  0.8.0                hd77b12b_0  
zlib                      1.2.13               h8cc25b3_0  
zstandard                 0.19.0          py312h2bbff1b_0  
zstd                      1.5.5                hd43e919_0  

```

gitterman, 2025-02-09:
I tried to install the latest rev for windows https://github.com/CadQuery/CQ-editor/releases/download/nightly/CQ-editor-master-Windows-x86_64.exe for all users, but it always failed in the end with a message that it couldn't link something.
The installer does not allow re-installing, if the installation directory isn't empty.
So I deleted the installation directory manually and installed again.
This time I did not double-click the exe from file explorer, but ran it from an administrative command prompt instead.
The installation was hanging at 'Setting up the base environment' when windows came up with the 'CQ-editor Installer has stopped working'

[image]
[image]

gitterman, 2025-02-09:
I did another attempt from explorer by right-clicking the installer to 'run as admin'.
This time it went a little further but eventually failed again.

[image]