CadQuery/cadquery #1784: Interactive show

adam-urbanczyk, 2025-03-05:
Currently `show` is blocking, it would be great to be able to use it multiple times without blocking of the script execution. I see 3 options

1. Use `Qt` or `Tk`, requires integration of the gui event loop with the interpreter (Check how IPython odes that)
2. Run all the VTK stuff in a separate thread and periodically invoke `vtkRenderWindowInteractor.ProcessEvents()`
3. Trame+pywebview

adam-urbanczyk, 2025-03-08:
Regarding 3, it is easy to get something quickly:

[image]

Few observations:

* `vtkAssembly` does not seem to work with vtkLocalView (VTK.js)
* As-is I'm getting too much output in the console in Spyder
```
Output from spyder call 'get_state':
vtk-remote-view.size is missingvtk-remote-view.size is missingvtk-remote-view.shape is missingvtk-remote-view.shape is missingvtk-remote-view.shape is missingvtk-remote-view.shape is missingvtk-remote-view.size is missingvtk-remote-view.size is missingvtk-remote-view.size is missingvtk-remote-view.size is missingvtk-remote-view.shape is missingvtk-remote-view.shape is missingvtk-remote-view.shape is missingvtk-remote-view.shape is missingvtk-remote-view.size is missingvtk-remote-view.size is missing
```

lorenzncode, 2025-03-08:
I had some interest in a VTK.js based viewer (maybe as a separate project not built-in to CadQuery).  I suppose the viewer could be implemented in other frameworks so Trame is one option.  

3. VTK.js single page application (browser and/or webview)
  a. Trame
  b. from scratch (bring your own framework)

adam-urbanczyk, 2025-03-09:
FYI: with the current implementation VTK.js based rendering seems to be rather quirky, less so with remote rendering. With VTK 9.4 where is also a WASM option if I understand [correctly](https://github.com/Kitware/trame-vtklocal), but for now we are on 9.3.

More visuals:
[image]

PR #1786 (2025-07-30): WIP: non-blocking show using trame/vtk
Closes #1784 (?)

API:
```python
from cadquery.fig import Figure
from cadquery.func import sphere, torus

# this starts a server and opens a browser tab
fig = Figure()

# we can use show, the API is fluent
fig.show(torus(10, 2), color='violet').show(sphere(5))

# zoom to fit
fig.fit()

# clear
fig.clear()
```

[image]