CadQuery/cadquery #1791: Inconsistent behavior when displaying objects with show_object()

lukman06, 2025-03-20:
## Description
There appears to be inconsistent behavior when displaying a Sketch object directly with `show_object()`. The function seems to automatically assemble the Sketch before displaying it.

## Code to reproduce
```python
import cadquery as cq
line = cq.Sketch().segment((0,0),(0.25,1)).segment((1,2))
line_assemble = line.assemble()
show_object(line)  # This shows the assembled version despite 'line' not being assembled
```

## Expected behavior
I would expect `show_object(line)` to either show the unassembled Sketch object or require explicit assembly before displaying.

## Actual behavior
When calling `show_object(line)`, it displays the Sketch as if `line_assemble = line.assemble()` had been called first. This automatic conversion happens behind the scenes.

## Environment
- CQ-Editor version: 0.5.dev0

lorenzncode, 2025-03-21:
Try copy:
```py
import cadquery as cq
import copy

sk1 = cq.Sketch().segment((0, 0), (0.25, 1)).segment((1, 2))
sk2 = copy.copy(sk1)
sk1.assemble()
```

or using `compound`:

```py
import cadquery as cq
from cadquery.func import compound
from cadquery.vis import show

sk1 = cq.Sketch().segment((0, 0), (0.25, 1)).segment((1, 2))
comp1 = compound(sk1.vals())
sk1.assemble()

show(comp1)
```

lorenzncode, 2025-03-21:
You could also visualize the list of edges directly in CQ-Editor or with cadquery.vis:

```py
sk1 = cq.Sketch().segment((0, 0), (0.25, 1)).segment((1, 2))
edges_list = sk1.vals()
sk1.assemble()
show_object(edges_list)
```