CadQuery/cadquery #1793: Imprinted geometry has more parts than original geometry

davidepettinari, 2025-03-25:
While imprinting a cad geometry I notice the number of imprinted parts is larger than the number of parts in the original geometry

we have 5 parts in the original step file and 9 in the imprinted geometry

step file and python script attached that reproduces the issue

I would have exspected the imprinted geometry to also have 5 parts like the original

[oktavian_a_info.zip](https://github.com/user-attachments/files/19450896/oktavian_a_info.zip)

```python
import cadquery as cq

parts = cq.importers.importStep("oktavian_a_info.step").val()

assembly = cq.Assembly()
for part in parts:
    print('orignal part', part)
    assembly.add(part)

imprinted_assembly, imprinted_solids_with_org_id = cq.occ_impl.assembly.imprint(assembly)
print()

for ipart in imprinted_assembly:
    print('imprinted part', ipart)
```

tagging @eepeterson @SteSeg @shimwell

davidepettinari, 2025-03-25:
Just realised that we might have parts that overlap in the same space.

I shall continue looking into this

adam-urbanczyk, 2025-03-25:
Your model is invalid according to OCCT (and by eye). You might want to reconsider how you make it.

```python
from cadquery.func import check

check(parts)
```

davidepettinari, 2025-03-27:
The CAD model was created in SpaceClaim — is there a recommended procedure to export a STEP file that's fully compatible with OCCT?

adam-urbanczyk, 2025-03-28:
AFAIC your model is non-manifold and you need to fix that. Do you recognize that?