CadQuery/cadquery #1799: CoolProp fails to import when cadquery is imported first

mariap-pf, 2025-04-04:
When importing cadquery before CoolProp, the script silently hangs and does not proceed to execution. Reversing the import order resolves the issue. Does anyone have an idea of what could be the problem?
## To Reproduce

```python
import cadquery
import CoolProp

if __name__ == "__main__":
    print("hello world")
```

## Environment

OS: Ubuntu 22.04
Python version: 3.10
cadquery version: https://github.com/CadQuery/cadquery/tree/2698db5c395705eb36e42bb5d4ab5944daf24436
CoolProp version: 6.6.0
Was CadQuery installed using Conda?: no, it was installed using pip
Using: Python interpreter

adam-urbanczyk, 2025-04-04:
Coolprop is not related in any way to CQ so that is rather esoteric (guess: ABI issue, but I wouldn't know how/why). Would you be able to:

a) check with conda-forge built packages (just download micromamba and do not bother with setting up your shell) 
b) attach `gdb` and provide a backtrace `bt`

lorenzncode, 2025-04-06:
I can reproduce the hang without cadquery.  I'll append to https://github.com/CoolProp/CoolProp/issues/2512

eguiraud-pf, 2025-08-01:
Added some more details and a proposed fix at https://github.com/CoolProp/CoolProp/issues/2512#issuecomment-3141629917 (namely, using latest master for the C++ rapidjson version in coolprop).

By sheer luck the C++ rapidjson version that cadquery's wheels use seems to be compatible with the latest C++ rapidjson commit, even if the cadquery-ocp build system does not really pick a version: if I understand correctly, it just looks for rapidjson in the system.

@adam-urbanczyk I would recommend to patch cadquery-ocp's build system so that it fetches the latest version of C++ rapidjson rather than picking it up from the system.

eguiraud-pf, 2025-08-01:
I don't think we are in the position to switch from PyPI to conda any time soon at Proxima, but point taken that this is likely a non-issue in the conda world.