CadQuery/cadquery #1817: Wrong definition of the Compound

XeuTap, 2025-04-24:
The documentation says that Compound is "a collection of disconnected solids". But this is not the exact definition. If we refer to the OpenCascade documentation, we can find the following sentences:
"COMPOUND: A group of any of the shapes below."
[TopAbs_ShapeEnum Description](https://dev.opencascade.org/doc/refman/html/_top_abs___shape_enum_8hxx.html#a67b8aa38656811eaee45f9df08499667)
"A TCompound is an all-purpose set of Shapes"
[TopoDS_TCompound Description](https://dev.opencascade.org/doc/refman/html/class_topo_d_s___t_compound.html#details)

This inaccuracy leads for the unexpected behavior.

### Examples:

### Compound's center of mass
```python
sk = cadquery.Sketch().rect(50, 50).faces()
face1 = sk.val()
face2 = face1.copy().translate(Vector(100, 0, 0))
compound = cadquery.Compound.makeCompound([face1, face2])
compound_center = compound.Center()
Properties = GProp_GProps()
BRepGProp.SurfaceProperties_s(compound.wrapped, Properties)
expected_center = cadquery.Shape.CombinedCenter([face1, face2]) # Vector: (50, 0.0, 0.0)
actual_center = compound.Center() # Vector(0.0, 0.0, 0.0)

assert actual_center  == expected_center, "Incorrect center of mass of the compound, expected {}, got {}".format(expected_center, compound.Center())
``` 

The following code makes a compound consisting of 2 faces and calculates the center of mass. The problem is that `shape_properties_LUT` interpets `TopAbs_COMPOUND` as a volumetric shape and provides an incorrect result.

### Shell
Similar behaviour occurs in the Shell.makeShell method

```python
sk = cadquery.Sketch().rect(50, 50).faces()
face1 = sk.val()
face2 = face1.copy().translate(Vector(50, 0, 0))
face3 = face1.copy().translate(Vector(100, 0, 0))
shell_connected = cadquery.Shell.makeShell([face1, face2]) # TopoDS_Shell
assert isinstance(shell_connected.wrapped,TopoDS_Shell)
shell_disconnected = cadquery.Shell.makeShell([face1, face3]) # TopoDS_Compound
assert isinstance(shell_disconnected.wrapped, TopoDS_Shell), "Wrapped type mismatch, expected TopoDS_Shell, got {}".format(shell_disconnected.wrapped)
``` 

This code creates 2 shells: one from the connected faces, and another from disconnected. The type of the second shell is `TopoDS_Compound`, which corresponds to the OpenCascade definition. However, cadquery states that the wrapped attribute of the shell object should be `TopoDS_Shell`. If we call `Center()`, we'll get the same behaviour as in the previous example.

If this feedback is relevant, I’d be happy to open a PR with a solution that addresses the specific issue with Center(). However, I’m unsure whether other operations may also be affected by this underlying inconsistency.

adam-urbanczyk, 2025-04-24:
Thanks, the `Center()` thing is definitely a bug. Regarding shell, I'd suggest using `shell` from the free function API. Parts of the `cq.Shape` API are rather dusty, and not really meant for end users.

XeuTap, 2025-04-25:
Thank you, @adam-urbanczyk. I opened a PR where I reimplemented `Center ` for `cq.Compound`.  There is also a utility method that that inspects the `Compound`.
#1822