CadQuery/cadquery #1827: STEP to GLTF: auto rotate x=90 degrees. Why?

kalentivan, 2025-04-27:
Hi, i have been using CadQuery  for the past year, and the STEP-to-GLTF conversion has consistently worked correctly with matching orientations between the two formats. However, over the last month, I noticed that during the conversion process, the model is now being rotated by 90 degrees around the X-axis. This issue started recently, even though there were no changes to my code or workflow. Now use 2.5.2. Why?

Steps to Reproduce:
Load a STEP file using CadQuery.
Export the model to GLTF format.
Observe that the orientation of the model in the GLTF file is rotated by 90 degrees around the X-axis compared to the original STEP file.

Environment:
CadQuery Version: 2.5.2
Python Version: [3.12 Python version]
Additional Dependencies: [trimesh]

Thank you!

adam-urbanczyk, 2025-04-29:
No clue to be honest. It might be a bug related to a dep of CQ. Do you have a MRE?

kalentivan, 2025-05-09:
Good afternoon, I apologize for the delayed response.

I've created an MRE (Minimal Reproducible Example) project here:
https://gitflic.ru/project/kalent/mre_cq

It includes my previous GLTF model from January 21st and a file with the model I converted now.
The STEP file itself hasn't changed. I also added a top-view projection in SVG to verify that the model is indeed positioned correctly. Previously, the model was oriented the same way, so the orientation in the STEP file definitely remained unchanged.

I just checked in the standard Windows viewer, and now it seems to generate GLTF correctly—that is, it orients the model along the Z-axis, whereas before it was oriented along the X-axis. But again, for some reason, it now orients them differently than before.

In my project, I render GLTF on the frontend, and that's where I noticed that the models started displaying differently. At first, I thought maybe something had changed on the frontend, but if that were the case, all previously generated models would have started rendering differently. However, this issue only occurs with newly created ones. Additionally, no one has made any changes to the frontend in a long time.

I’ve also attached a file with all the dependencies in the project—perhaps the issue stems from one of them.