CadQuery/cadquery #1834: How to constrain an object with an object added to a different assembly

Vishnu-C, 2025-05-03:
```
import cadquery as cq

# create an assembly 
assmebly1 = cq.Assembly(name="assembly1")
# add box to assembly1
box = cq.Workplane().box(1, 1, 1)

assmebly1.add(box, name="box1", color=cq.Color("green"))
assmebly1.add(box, name="box2", color=cq.Color("yellow"))

assmebly1.constrain("box1@faces@<X", "box2@faces@<X", "Axis", param=0.0)
assmebly1.constrain("box1@faces@<Y", "box2@faces@<Y", "Axis", param=0.0)
assmebly1.constrain("box1@faces@<Z", "box2@faces@<Z", "PointInPlane", param=10.0)
assmebly1.solve()

assmebly2 = cq.Assembly(name="assembly2")
# add box to assembly2
assmebly2.add(box, name="box3", color=cq.Color("purple"))

# -----> how to constrain box3 in assembly2 with box1 from assembly1?
# I should not add box1 to assembly2

# assmebly2.solve()
```

lorenzncode, 2025-05-03:
Related: https://github.com/CadQuery/cadquery/issues/1239

It is possible to reference the location of the object in the other assy: 

```py
assembly2.add(box, name="box3", loc=assembly1.objects['box2'].loc)
```

adam-urbanczyk, 2025-06-03:
In general you cannot reference unrelated assembly. The example above will not update if assy1 changes. You could add assy1 to assy2 as a sub-assembly.

Vishnu-C, 2025-06-08:
For my use case I need the object only to be referred and not add to the current assembly. Did a trick by adding the object with its solved location, solve the current assembly and use remove() function, which fortunately does not re-solve constrains after remove.

assembly2.add(box, name="box3", loc=assembly1.objects['box2'].loc)
// add constraints
...
assembly2.solve()
assembly2.remove("box3")

Cheers!