CadQuery/cadquery #1841: Wire's offset2D can work as expected when d > 0, and cannot work as expected when d < 0

huskier, 2025-05-15:
When d > 0, the following code can work as expected.
```
import cadquery as cq
from cadquery.vis import show
from cadquery.func import *

result1 = rect(20, 20)
result2 = wire(result1)
result3 = result2.offset2D(d=5.0, kind = "arc")
result4 = extrude(face(result1), (0, 0, 1)).translate((0, 0, 1))
result5 = extrude(face(result3), (0, 0, 1))
show(result4+result5)
```

However, when d < 0, the following code can not work as expected. and the result5's corners should be rounded.
```
import cadquery as cq
from cadquery.vis import show
from cadquery.func import *

result1 = rect(20, 20)
result2 = wire(result1)
result3 = result2.offset2D(d=-2.0, kind = "arc")
result4 = extrude(face(result1), (0, 0, 1))
result5 = extrude(face(result3), (0, 0, 1)).translate((0, 0, 1))
show(result4+result5)
```

adam-urbanczyk, 2025-05-15:
Actually it works as expected AFAICT. The corners cannot be rounded for a negative offset wire. `arc` is for something else - approximation of the wire to lines+arcs before offsetting.

huskier, 2025-05-15:
Thank you for your explanation. I am thinking about it.

closed 2025-05-15 by huskier