CadQuery/cadquery #1854: Segmentation fault in GitLab CI (Linux/amd64)

dAnjou, 2025-06-03:
Hi there 👋 

Locally, on my Apple M1 Pro, everything works fine. But on GitLab CI there is a segmentation fault.

_Sidenote: #1475 doesn't seem to be an issue anymore, CQ installs fine in both environments mentioned above._

## To Reproduce

Here's my CI build with the segmentation fault: https://gitlab.com/dAnjou/bed/-/jobs/10236020027
The README contains instructions: https://gitlab.com/dAnjou/bed

_Just to avoid confusion: the current build in https://gitlab.com/dAnjou/bed/-/tree/55ffbafb0c0a3b4ed01cb65c04b60ae1b0eaebd9/public/bed is from my local machine, not the CI build._

## Environment

OS: `python:3.12` Docker container on Linux/amd64
CQ version: `git+https://github.com/CadQuery/cadquery.git`

---
Sorry in advance for the code 😅  I'm not familiar with CAD concepts and can't wrap my head around some of them. Happy to provide more info, let me know what you need.

dAnjou, 2025-06-03:
I have added a `Dockerfile` which reproduces the issue - container exits with SIGSEGV - when I run it in this colima setup:
```
PROFILE    STATUS     ARCH      CPUS    MEMORY    DISK      RUNTIME    ADDRESS
default    Running    x86_64    4       4GiB      100GiB    docker
```

dAnjou, 2025-06-03:
Updated the Docker setup: https://gitlab.com/dAnjou/bed/-/commit/33cc01e1df55afd381c90c6fcca8fd89166090bb?view=parallel, running `docker run bed python bed.py` still results in a SIGSEGV.

lorenzncode, 2025-06-04:
I ran the container locally and can reproduce it.  Here is the backtrace after installing gdb.

```
podman run --rm -it cq1854 gdb -ex r --args python bed.py
...
```

```
#0  0x00007faad0d38bc9 in XSync () from /opt/conda/lib/python3.12/site-packages/../.././libX11.so.6
#1  0x00007faaced0b6bd in vtkXRenderWindowInteractor::Initialize() () from /opt/conda/lib/python3.12/site-packages/../../././libvtkRenderingUI-9.3.so.1
#2  0x00007fa9b6ea97ed in PyvtkRenderWindowInteractor_Initialize(_object*, _object*) () from /opt/conda/lib/python3.12/site-packages/vtkmodules/vtkRenderingCore.cpython-312-x86_64-linux-gnu.so
#3  0x0000555efba61d59 in cfunction_call (func=0x7fa9b047b1a0, args=<optimized out>, kwargs=<optimized out>) at /usr/local/src/conda/python-3.12.10/Objects/methodobject.c:548
#4  0x0000555efba4526c in _PyObject_MakeTpCall (tstate=0x555efbecde90 <_PyRuntime+458992>, callable=0x7fa9b047b1a0, args=0x7faae144f308, nargs=<optimized out>, keywords=0x0)
    at /usr/local/src/conda/python-3.12.10/Objects/call.c:240
#5  0x0000555efb958b4f in _PyEval_EvalFrameDefault (tstate=<optimized out>, frame=0x7faae144f160, throwflag=<optimized out>) at Python/bytecodes.c:2715
#6  0x0000555efbaf64f1 in PyEval_EvalCode (co=co@entry=0x555f02560730, globals=globals@entry=0x7faae15ad180, locals=locals@entry=0x7faae15ad180)
    at /usr/local/src/conda/python-3.12.10/Python/ceval.c:580
#7  0x0000555efbb30afa in run_eval_code_obj (tstate=tstate@entry=0x555efbecde90 <_PyRuntime+458992>, co=co@entry=0x555f02560730, globals=globals@entry=0x7faae15ad180,
    locals=locals@entry=0x7faae15ad180) at /usr/local/src/conda/python-3.12.10/Python/pythonrun.c:1757
#8  0x0000555efbb2b7b5 in run_mod (mod=mod@entry=0x555f025cb650, filename=filename@entry=0x7faae135b0b0, globals=globals@entry=0x7faae15ad180, locals=locals@entry=0x7faae15ad180,
    flags=flags@entry=0x7ffd7a1e5e20, arena=arena@entry=0x7faae14d3c50) at /usr/local/src/conda/python-3.12.10/Python/pythonrun.c:1778
#9  0x0000555efbb28850 in pyrun_file (fp=fp@entry=0x555f02512490, filename=filename@entry=0x7faae135b0b0, start=start@entry=257, globals=globals@entry=0x7faae15ad180,
    locals=locals@entry=0x7faae15ad180, closeit=closeit@entry=1, flags=0x7ffd7a1e5e20) at /usr/local/src/conda/python-3.12.10/Python/pythonrun.c:1674
#10 0x0000555efbb2856e in _PyRun_SimpleFileObject (fp=0x555f02512490, filename=0x7faae135b0b0, closeit=1, flags=0x7ffd7a1e5e20) at /usr/local/src/conda/python-3.12.10/Python/pythonrun.c:459
#11 0x0000555efbb282a4 in _PyRun_AnyFileObject (fp=0x555f02512490, filename=filename@entry=0x7faae135b0b0, closeit=closeit@entry=1, flags=flags@entry=0x7ffd7a1e5e20)
    at /usr/local/src/conda/python-3.12.10/Python/pythonrun.c:78
#12 0x0000555efbb251c2 in pymain_run_file_obj (skip_source_first_line=0, filename=0x7faae135b0b0, program_name=0x7faae135b0f0) at /usr/local/src/conda/python-3.12.10/Modules/main.c:361
#13 pymain_run_file (config=0x555efbe70a70 <_PyRuntime+77008>) at /usr/local/src/conda/python-3.12.10/Modules/main.c:380
#14 pymain_run_python (exitcode=0x7ffd7a1e5df4) at /usr/local/src/conda/python-3.12.10/Modules/main.c:634
#15 Py_RunMain () at /usr/local/src/conda/python-3.12.10/Modules/main.c:714
#16 0x0000555efbae05d7 in Py_BytesMain (argc=<optimized out>, argv=<optimized out>) at /usr/local/src/conda/python-3.12.10/Modules/main.c:768
#17 0x00007faae163f24a in __libc_start_call_main (main=main@entry=0x555efbae0520 <main>, argc=argc@entry=2, argv=argv@entry=0x7ffd7a1e6078) at ../sysdeps/nptl/libc_start_call_main.h:58
#18 0x00007faae163f305 in __libc_start_main_impl (main=0x555efbae0520 <main>, argc=2, argv=0x7ffd7a1e6078, init=<optimized out>, fini=<optimized out>, rtld_fini=<optimized out>,
    stack_end=0x7ffd7a1e6068) at ../csu/libc-start.c:360
#19 0x0000555efbae047d in _start ()
```

The problem is in generating the screenshots with `cadquery.vis.show`.

dAnjou, 2025-06-04:
Thank you both very much! 🙏 

I can sidestep the problem by exporting SVGs. But I will give `xvfb` a shot as well 👍

NothanUmber, 2025-06-05:
Not sure whether this is related, getting the following problem on a Macbook Pro M2:

% poetry show cadquery          
 name         : cadquery                                                                            
 version      : 2.5.2                                                                               
 description  : CadQuery is a parametric  scripting language for creating and traversing CAD models 

dependencies
 - cadquery-ocp >=7.7.0,<7.8
 - casadi *
 - ezdxf *
 - multimethod >=1.11,<2.0
 - nlopt >=2.9.0,<3.0
 - path *
 - typish *
% python --version              
Python 3.11.11
% python                        
Python 3.11.11 (main, May 26 2025, 14:49:42) [Clang 17.0.0 (clang-1700.0.13.3)] on darwin
Type "help", "copyright", "credits" or "license" for more information.
>>> import cadquery as cq
zsh: segmentation fault  python

dAnjou, 2025-06-05:
Made some progress: https://gitlab.com/dAnjou/bed/-/commit/9be86dd49ff344915328345e52a101d3bbe2dd07 (sorry, many changes, most relevant are the ones in `Dockerfile` and `.gitlab-ci.yml`).

I have integrated `xvfb`, and the segmentation fault is gone when exporting screenshots 🥳  Unfortunately though, the screenshots (linked in the [README](https://gitlab.com/dAnjou/bed/-/blob/main/README.md)) are mostly black except for the axes indicator 😢

I've talked to ChatGPT a bit, and it suggested installing Mesa dependencies, so I added `libglx-mesa0` (which also brings `libgl1-mesa-dri` as before).

I also asked it how I can check whether things generally work. It suggested doing these three things inside the container, which I did:
```shell
glxinfo | grep "OpenGL"
```
Shows OpenGL available ✅ 
```python
import vtk

rw = vtk.vtkRenderWindow()
rw.SetOffScreenRendering(1)
rw.Render()  # Try rendering

print("Offscreen rendering is:", rw.GetOffScreenRendering())
```
Prints `Offscreen rendering is: 1`, and doesn't crash ✅ 
```python
import vtk

# Create a simple renderer, render window, and interactor
renderer = vtk.vtkRenderer()
render_window = vtk.vtkRenderWindow()
render_window.AddRenderer(renderer)
render_window.SetOffScreenRendering(1)  # Enable offscreen

# Example actor (a cone)
cone = vtk.vtkConeSource()
mapper = vtk.vtkPolyDataMapper()
mapper.SetInputConnection(cone.GetOutputPort())
actor = vtk.vtkActor()
actor.SetMapper(mapper)
renderer.AddActor(actor)

# Render and capture image
render_window.Render()
window_to_image = vtk.vtkWindowToImageFilter()
window_to_image.SetInput(render_window)
window_to_image.Update()

writer = vtk.vtkPNGWriter()
writer.SetFileName("screenshot.png")
writer.SetInputConnection(window_to_image.GetOutputPort())
writer.Write()
```
Produces a screenshot with a visible cone ✅

dAnjou, 2025-06-06:
Thank you again, @lorenzncode!

I've forked CQ and applied your change: https://github.com/dAnjou/cadquery/commit/c4b51833f2e33c0397f287fa572dbfd1ca33625b

Using that in my project now: https://gitlab.com/dAnjou/bed/-/commit/1bea95a0d3022f677ffe6a1a262005ab16427ab0

And the screenshots are fixed now: https://gitlab.com/dAnjou/bed/-/blob/main/README.md (might have to clear browser cache to see them) 🥳

closed 2025-11-08 by jmwright