CadQuery/cadquery #1866: Sweep missing surfaces in rounded corners

JoshuaDietz, 2025-07-06:
I have the following cadquery code:

```python
def container_base_profile():
    width = 42.5
    half_width = width / 2
    profile = cq.Workplane("XY").polyline([(0, 0), (half_width - 2.95, 0), (half_width - 2.15, 0.8), (half_width - 2.15, 2.6), (half_width, 4.75), (half_width, 5), (0, 5)]).close()
    
    temp_box = (cq.Workplane("XZ")
                .rect(42.5, 42.5)
                .extrude(1)  
                .edges("|Y") 
                .fillet(7.5))

    top_view = temp_box.faces(">Y").wires().val()

    result = profile.sweep(top_view)

    return result

show_object(container_base_profile())
```

Which produces the following result

[image]

As you can see, the result looks good **except for the missing faces in the corners**.
Does anyone have an idea how to fix these faces? 
I've read multiple times that its usually better to use a loft instead. I will try that next but I'm still curious if there's a solution for the missing faces. I've already tried:

- `profile.sweep(top_view, makeSolid=True)`
- `profile.sweep(top_view, isFrenet=True)`
- `result = result.shell(0.1).shell(-0.1)`

All without any success.

Additionally I'd be intrested in other suggestions to make the path to sweep along. Doing the extrude and then choosing the wires of the face seems overly complicated, but after hours of trial & error this was the only working solution that I could come up with

For reference here's a picture of the sketch and the path along which the sketch is swept:

[image]

Thank you for your help :)

lorenzncode, 2025-07-06:
The solid might be constructed using split by creating a shell from the sweep using an open profile instead of closed.:

```py
import cadquery as cq

width = 42.5
half_width = width / 2
h = 4.75

# create the wire profile
profile = (
    cq.Workplane("XY")
    .polyline(
        [
            (half_width - 2.95, 0),
            (half_width - 2.15, 0.8),
            (half_width - 2.15, 2.6),
            (half_width, h),
        ]
    )
    .wire()
)
# 2D sketch of shape without the profile
sk1 = cq.Sketch().rect(width, width).vertices().fillet(7.5)
# create solid
solid1 = cq.Workplane("ZX").placeSketch(sk1).extrude(h)
# the path for the sweep
top_view = solid1.faces(">Y").wires().val()
# shell result of the sweep
profile_shell = profile.sweep(top_view)
# split the solid, sort by Volume to select the desired result
result = solid1.split(profile_shell).solids().sort(lambda s: s.Volume())[-1]
```

lorenzncode, 2025-07-06:
> Additionally I'd be intrested in other suggestions to make the path to sweep along. Doing the extrude and then choosing the wires of the face seems overly complicated

Here is another way to do it without the extrude.  This creates the solid from shells.

```py
import cadquery as cq
from cadquery.func import *

width = 42.5
half_width = width / 2
h = 4.75

# create the wire profile
profile = (
    cq.Workplane("XY")
    .polyline(
        [
            (half_width - 2.95, 0),
            (half_width - 2.15, 0.8),
            (half_width - 2.15, 2.6),
            (half_width, h),
        ]
    )
    .wire()
)
# 2D sketch of shape without the profile
# reset() because we are done with the vertices selection
sk1 = cq.Sketch().rect(width, width).vertices().fillet(7.5).reset()
# the path for the sweep
top_view = sk1.wires().val().move(rx=90, y=h)
# shell result of the sweep
profile_shell = profile.sweep(top_view).val()
# create solid from shells
top_face = face(wire(profile_shell.edges(">Y")))
bottom_face = face(wire(profile_shell.edges("<Y")))
result = solid(shell(bottom_face, profile_shell.faces(), top_face))
```

Here is a another version replacing `Workplane`, `Sketch` usage with the free function API.

```py
from cadquery.func import *

width = 42.5
half_width = width / 2
h = 4.75

# create the wire profile
pts = [
    (half_width - 2.95, 0),
    (half_width - 2.15, 0.8),
    (half_width - 2.15, 2.6),
    (half_width, h),
]
profile = polyline(*pts)
# the path for the sweep
wire1 = rect(width, width)
top_view = wire1.fillet2D(7.5, wire1.vertices()).move(rx=90)
# shell result of the sweep
profile_shell = sweep(profile, top_view)
# create solid from shells
top_face = face(wire(profile_shell.edges(">Y")))
bottom_face = face(wire(profile_shell.edges("<Y")))
result = solid(shell(bottom_face, profile_shell.faces(), top_face))
```

JoshuaDietz, 2025-07-06:
Thank you so much! I tried all out, all are working like a charm :)

closed 2025-07-06 by JoshuaDietz