CadQuery/cadquery #1886: Zenodo DOI still points to CadQuery 2.4.0 instead of latest 2.5.2

huskier, 2025-08-22:
I noticed that the Zenodo DOI listed for scientific citation — https://doi.org/10.5281/zenodo.3955118 — still resolves to the CadQuery 2.4.0 release. However, the latest version is 2.5.2, and it would be great if the DOI could be updated to reflect the most recent release.

[image]

This ensures that researchers citing CadQuery are referencing the latest features and improvements.

Thanks for the work on this excellent project!

jmwright, 2025-10-07:
I cannot remember the steps for updating this, or if I have ever done it. @adam-urbanczyk If you can list the general steps, I can try to update it and will write a developer doc.

adam-urbanczyk, 2025-10-28:
FWIW 2.6.1 is visible.

huskier, 2025-10-29:
Cheers! 

Thank you very much.