CadQuery/cadquery #1891: When fusing two Faces, in some cases the internal edges cannot be removed.

huskier, 2025-08-30:
The following code can successfully fuse two Faces and remove the internal edges;
```
from cadquery.hull import find_hull
from cadquery.func import *
from cadquery.vis import show

base_face = face(rect(w=64, h=75, centered=True))
print(type(base_face))
show(base_face)

moved_face = face(rect(40, 40).moved(Location(44, 0, 0)))

combined = base_face + moved_face
print(type(combined))

combined = combined.clean()
show(combined)
```

however, the code below cannot remove the internal edges.
```
from cadquery.hull import find_hull
from cadquery.func import *
from cadquery.vis import show

base_face = face(rect(w=64, h=75, centered=True))

print(type(base_face))
show(base_face)

moved_square = rect(40, 40).moved(Location(44, 0, 0))
moved_circle = circle(20).moved(Location(82, 0, 0))

hull_shapes = moved_square + moved_circle
hull_edges = hull_shapes.Edges()
hull_wire = find_hull(hull_edges)
hull_face = face(hull_wire)

print(type(hull_face))
show(hull_face)

#combined_face = base_face + hull_face
combined_face = base_face.fuse(hull_face)
combined_face = combined_face.clean()

print(type(combined_face))
show(combined_face)
```

The difference between the two pieces of code is that in the working case, a base_face is fused with a simple rectangular face, while in the non-working case, the base_face is fused with a face obtained through a hull operation.

## Environment
We are on Windows 11 with CadQuery 2.4.0 under Anaconda environment.

jmwright, 2025-08-30:
@huskier Have you tried this in CadQuery 2.5.2?

huskier, 2025-08-30:
@jmwright 

I've just tried the example in CadQuery 2.5.2.

In CadQuery 2.5.2, to make the hull work properly, we have to do some modifications metioned in PR #1889.